OpenEVSE/openevse_esp32_firmware #224: OCPP does not report correct energy if EV is not unplugged

glynhudson, 2021-07-27:
If an OCPP session is stopped but the EV remains plugged in the wh reported by the EVSE does not increment, since the EVSE only increments lifetime wh when the EV is unplugged

[image]

@jeremypoulter @lincomatic is there a reason why the EVSE only increments lifetime wh when the EV is unplugged? I think it would be better if lifetime wh was incremented in real-time during a charging session

jeremypoulter, 2021-07-27:
Think this has always been the behaviour of the OpenEVSE. We can easily sum the total + session values if needed if the session value  clears when the total increases will have to do some tests.

Also on the subject, calling these wh/watthour and wattsecond is terrible, what do folks think about correcting the names and units in the v4 (breaking backwards compatible) ?

matth-x, 2021-11-04:
This is addressed in #255

@glynhudson Can you ckeck if the PR solves this issue? Unfortunately, I cannot test it with a real load.

jeremypoulter, 2021-11-04:
@matth-x did you change this just for OCP we need to be careful about what is fixed in OCPP and what is at a higher level.

matth-x, 2021-11-04:
Yes I've only worked on the OCPP side of this.

jeremypoulter, 2021-11-04:
Then I think we should remove from OCPP, and fix generically for consistency.

matth-x, 2021-11-05:
Yes, that would be great. The energy reading is fully encapsulated in my code, so switching to another function will be no effort.

KipK, 2023-03-01:
PR https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/556 will breaks things here as now counters will be updated in real time
@matth-x I let you check the changes involed for OCPP ?

matth-x, 2023-03-01:
@KipK Thanks for the notice, I checked it and commented in the PR.

closed 2023-03-31 by glynhudson