OpenEVSE/openevse_esp32_firmware #225: Add SoC limit

glynhudson, 2021-07-28:
Now SoC integration is implemented it, SoC limit feature can now be implemented. The easiest way is probably a dropdown box under the energy limit with a few presets e.g 60%, 70%, 80%, 90%

kallisti5, 2021-08-07:
this would be nice... especially as a Bolt EV owner ;-) 
Tired of hopping into the car to tune the charge limit.

chris1howell, 2021-08-07:
Additionally a limit should be added for distance miles/km. Example... Charge to 300km range.

pdhoogh, 2022-05-20:
@chris1howell : as range is from GOM (_Guess_-O-Meter), the charge too will be a guess. Depends totally on load, weather, style... results from the past do not guarantee similar results in future etc... so, what are you precisely looking for?

KipK, 2023-01-31:
My 2 cents on this one. New Limit engine now handles time, energy, soc, and range.

closed 2023-02-06 by jeremypoulter

KipK, 2023-02-06:
Also added System Limits , persistant ones defined from config. 
UI is all done, please give me your feedback on the UX.  Enjoy :)

PR #535 (2023-02-06): New Limit engine for Wifi module. 
Limit engine for Wifi module. Replace limits from OpenEvse module.

New /limit API endpoint
Added classes:
Limit
LimitProperties
LimitType class

POST { "type": type, "value": value, "auto_release" : optional bool }
type actually possible are : 'time", "energy", "soc", "range" 

Disable state when time or energy session > defined limit.
Release disable claim when car is disconnected.

time_limit in minutes
session_energy in w/h

Todo: 
 - add configuration and feature on UI

fix #225