OpenEVSE/openevse_esp32_firmware #230: OCPP - unlock connector via command

matth-x, 2021-08-03:
In case the locking actuator of the plug fails, the CPO should be able to use the OCPP command `UnlockConnector` to trigger the EVSE to retry unlocking the connector. It is okay for the moment to reply with a `NotImplemented` message if the EVSE doesn't support it, but it would be really helpful to allow a more extensive remote management via OCPP.

jeremypoulter, 2021-08-03:
This would need some support from the OpenEVSE firmware, we started a discussion in https://github.com/lincomatic/open_evse/issues/141 but there was no conclusion. @chris1howell / @glynhudson  any thoughts?

lincomatic, 2021-10-29:
Hmm.. seems we need a RAPI lock/unlock command for Mennekes? If we do that, then I agree, we should just leave it to WiFi to control it. But if we're doing that, it really seems more logical to just take the EVSE out of the equation completely and connect the lock motor to the WiFi.
For non-Mennekes scenarios, the AUTH_LOCK code alone can handle the access control of charging

pavel167, 2021-11-03:
Hello,
Can someone suggest how unlock feature could realised on a charger, i.e. upon reception of "unlock connector" command over OCPP (from back-end), the charger should send some "unlock" command to connector over CAN bus ... something like this ?

chris1howell, 2021-11-03:
@pavel167

In Europe, it is common to bring your own cable, the cable has a connector
on both ends. The station locks the connector to the station when the
charge starts. Once the vehicle end is unlocked and unplugged the station
unlocks the station side connector.

OCPP connector unlock allows a command to unlock the station side connector.

jeremypoulter, 2021-12-24:
@matth-x this is now implemented in the OpenEVSE firmware, #289 covers adding the generic handling in the WiFi firmware

closed 2023-03-31 by glynhudson