OpenEVSE/openevse_esp32_firmware #231: CORS - OpenEVSE save broken when connecting using hostname

kwri-avongluck, 2021-08-03:
If you access the OpenEVSE unit via a hostname. saving settings fails due to CORS

Impacted versions:
* OpenEVSE 4.0.1
* OpenEVSE 4.1.0

[image]

The cause seems to be the request using the IP address of the unit directly instead of the hostname the unit was accessed via.

glynhudson, 2021-08-03:
I've been unable to recreate this when accessing the unit via local hostname e.g http://openevse-53ad/. It looks like the issue only effects remote hostnames? Thanks for reporting

jeremypoulter, 2021-08-03:
It theory this should only happen for .local addresses, because these can sometimes take a while to resolve, we switch to IP address, but that shouldn't be a problem as by default we allow requests from any URL. Will however take a look

kallisti5, 2021-08-07:
yeah, it's more of a browser side issue. Javascript on openevse.discord.local contacting an IP which isn't literally "openevse.discord.local" is generally guarded against as a cross-site scripting attack.

I should also mention, contacting the openevse unit via IP results in the issue going away :-)

jeremypoulter, 2021-08-30:
Can you give some more details on your setup? What browser? What is discord.local? how is that connected to openevse?

kallisti5, 2021-08-30:
* Every browser tested. (Firefox, Opera, Chrome)
* discord.local is just my local dnsmasq dns domain

```
$ ping openevse.discord.local
PING openevse.discord.local (192.168.1.221) 56(84) bytes of data.
64 bytes from openevse.discord.local (192.168.1.221): icmp_seq=1 ttl=255 time=102 ms
```

```
$ ping openevse
PING openevse.discord.local (192.168.1.221) 56(84) bytes of data.
64 bytes from openevse.discord.local (192.168.1.221): icmp_seq=1 ttl=255 time=61.6 ms
64 bytes from openevse.discord.local (192.168.1.221): icmp_seq=2 ttl=255 time=78.3 ms
```

* Accessing via IP works
* Accessing via Hostname doesn't work due to CORS.

Essentially, the javascript is accessing the local API via IP somehow instead of what the browser is connecting with (request.headers.host)

kallisti5, 2021-08-30:
I did a quick grep of openevse_wifi_gui...

```
src/view_models/OpenEvseWiFiViewModel.js:        if(self.baseHost().endsWith(".local") && "" !== self.status.ipaddress()) {
src/view_models/OpenEvseWiFiViewModel.js:            self.baseHost(self.status.ipaddress());
src/view_models/OpenEvseWiFiViewModel.js:            window.location.replace("http://" + self.status.ipaddress() + ":" + self.basePort());
```

It's doing a bunch of weird stuff trying to detect accessing via Hostnames ending in .local and redirecting to the IP address.  Not really sure why any of that logic exists :-)

You could test this all locally without a local network dns server by adding your openevse [to your local hosts file](https://www.howtogeek.com/howto/27350/beginner-geek-how-to-edit-your-hosts-file/):

evseIP   mycoolevse.network

jeremypoulter, 2021-08-30:
The `.local` domain is [reserved for mDNS](https://www.rfc-editor.org/rfc/rfc6762.html) so shouldn't really be used for normal DNS, not that that should have any bearing on this issue, but it is the reason for why we switch to IP address. It used to be the case that mDNS resolutions where slow so we switched to IP once we knew it.

This should however not be an issue and looks to be working fin more 'normal' mDNS resolved hosts:
[image]
You can see we are adding the CORS headers to allow connection from everywhere.

Did you disable the `enableCors` option? Do you have stricter browser settings? What response headers are being sent?

kallisti5, 2021-08-30:
Standard default browser settings.
It does look like this might be an interaction of mDNS vs .local dns code.  Wasn't aware of the .local reservation for mDNS, however things should still work the standard way instead of forcing the .local request over to an IP address. (I'm actually surprised those calls are working for you via mDNS) 

Looking at a simple query of the system page....

[image]

* The queries to "openevse.discord.local" are all fine and will pass CORS for GET/POST/PUT/etc
* The queries to the ip are also fine since they are all GET's and not posting data cross-site.

However, on saving settings, you're writing cross-origin to the ip instead of the hostname. This is when CORS kicks in.    Default settings on all browsers.  This test was done via firefox.

Clicking save...
[image]

As soon as the POST occurs to the IP CORS kicks in and blocks it.

kallisti5, 2021-08-30:
Crap.    I rechecked the browsers... I was mistaken on all browsers.

* Firefox 91.0.2 (64-bit)  - CORS block
* Opera 75.0.3960 - works
* Chrome 85.0.4183 - works

jeremypoulter, 2021-08-30:
`GET` should also fail if CORS is not working, certainly if I disable the sending of the CORS headers all requests to the IP are blocked for me, can you show me the response headers? Any more information in the console? eg this is what I get in Chrome (once I have forces a CORS error)
[image]

kallisti5, 2021-08-30:
console for the Firefox CORS error:
```
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://192.168.1.221/config. (Reason: CORS request did not succeed).
```

kallisti5, 2021-08-30:
Here's what Firefox documents for this error:
https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSDidNotSucceed

It mentioned " In many cases, it is caused by a browser plugin (e.g. an ad blocker or privacy protector) blocking the request. "  I connected via private mode, and still got the same issue (ruling out some local extension)

jeremypoulter, 2021-08-30:
Yeah, it does have the feeling of some extension causing this. That being said, I have been doing some testing and `.local` resolving looks to have improved a lot since that code was originally added so I think it is fairly safe to remove and disable CORS

PR #242 (2021-09-01): Remove CORS and associated `.local` redirect
Fixes #231
closed 2021-09-01 by glynhudson