OpenEVSE/openevse_esp32_firmware #232: Exposing the SoC?

kallisti5, 2021-08-07:
Does the latest OpenEVSE 4.1.0 expose or publish the vehicle's SoC anywhere?  Noticed the new "vehicle" tab, but it's usage is a little unclear.

Tried adding some MQTT topics to it, but the SoC, range, time to full, etc don't show up. The SoC also doesn't show up anywhere in the webui

```
openevse-unit/amp 30580
openevse-unit/voltage 240
openevse-unit/pilot 30
openevse-unit/wh 438626
openevse-unit/temp 502
openevse-unit/temp1 502
openevse-unit/temp2 false
openevse-unit/temp3 false
openevse-unit/temp4 507.5
openevse-unit/state 3
openevse-unit/vehicle 1
openevse-unit/colour 6
openevse-unit/manual_override 1
openevse-unit/freeram 208028
openevse-unit/divertmode 1
openevse-unit/srssi -64
```

chris1howell, 2021-08-07:
Other way around. The Station has no way to know SOC as the pilot is
analog. The new MQTT topics allow the station to recieved the SOC, range
and time to full via MQTT from an external source.

Once the station has the information it is displayed on the screen and new
timer options are possible. Charge to xxx miles/km of range... Charge to
xx% SOC.

Many vehicle APIs are unofficially documented, Python libraries are
avaliable for most vehicles.

kallisti5, 2021-09-01:
Ack.  I didn't realize the vehicles didn't offer SoC from vehicle to charger.  Guess I was thinking of DCFC which does have this ability.
Closing this one since it's beyond the scope of OpenEVSE.

closed 2021-09-01 by kallisti5