OpenEVSE/openevse_esp32_firmware #233: MQTT and Eco Mode

FCampos1977, 2021-08-11:
If we use MQTT to set Eco Mode, system works well, but i'm not able to see this in WebUI. 
Sometimes when we check WebUI, we found OpenEVSE in Normal Mode, but is in Eco Mode.
Reboot of Wifi module, solve the problem, but few days after is the same.

glynhudson, 2023-03-31:
I can confirm I'm able to replicate this bug. Enabling divert via MQTT e.g 

`<base-topic>divertmode/set` `2`

Results in the divert mode being successfully enabled, but the UI does not update

larroy, 2023-06-16:
I'm unable to set divert_active with GUI v2. I see in mqtt that divert_active is false. I haven't figured out how to set it in the GUI, the only way I made it work is with mqtt setting divert_active = true

larroy, 2023-06-16:
Using divertmode = 2 (Production) current is not modulated at all.

larroy, 2023-06-16:
[image][image]

larroy, 2023-06-16:
debug output shows:

```
{"divert_update":0,"solar":0,"charge_rate":0,"trigger_current":6,"voltage":240.7,"available_current":0,"smoothed_available_current":0,"pilot":25,"min_charge_end":0}

```

larroy, 2023-06-16:
Tried latest build from git b8d1addc4fcf47c51b1c559e5a10d9c0f382bfff and I made it work.

larroy, 2023-06-16:
Minimum charge time doesn't seem to be respected if smoothing and attack tau are 0

larroy, 2023-06-16:
I think there's a bug. If mqtt_solar is set, then grid_ie is not updated even if divert_type is DIVERT_TYPE_GRID

KipK, 2023-06-16:
Yep it was the case until recent dev builds. Previous divert config was set by matt topics so mqtt_solar needs to be empty if mqtt_grid is set.
UI was preventing doing this, but if you set using api it will be an issue. 
Now dev branch have a config flag for that and you can keep booth fields filled

larroy, 2023-06-17:
I compiled myself the latest master. I will do some changes that I can share. The current logic changes current too often.

jeremypoulter, 2023-06-17:
Not yet, there is however the divert_sim that can run various test cases. You will have to look at the GitHub action for build instructions. I have not had a chance to write more human friendly instructions.

closed 2026-09-21 by github-actions[bot]