OpenEVSE/openevse_esp32_firmware #234: Tesla account nod added

andriuskr, 2021-08-16:
Hello,

Trying to add tesla account (no MFA) to OpenEVSE. After saving credentials nothing happens, no info about Tesla SoC. {"count":0,"vehicles":[]}

jeremypoulter, 2021-08-16:
Yeah, was looking at this over the weekend, Tesla have changed their login code again and the library we use has not been updated.

andriuskr, 2021-08-20:
Is there any plans to update it ?

jeremypoulter, 2021-08-20:
For sure, waiting on a fix in the upstream project: mseminatore/TeslaJS#261

jeremypoulter, 2021-08-20:
Also will try to get #215 implemented so at least there is a work around.

andriuskr, 2021-11-06:
Where on OpenEVSE web i can enter tokens ?

glynhudson, 2021-11-07:
If you run the latest dev version, there is an advanced tab. 

[image]

Dev versions can now be downloaded from the releases page, marked as 'pre-release'. Be sure to select the correct release artefact to match your hardware.

andriuskr, 2021-11-07:
Update to newest dev version. Vehicle tab is now gone. 
[image]

chris1howell, 2021-11-07:
Try clearing browser cache.

andriuskr, 2021-11-07:
Tried even on new computer, not helping. Vehicle tab is gone

jeremypoulter, 2021-11-07:
Think this is related to a resent merge, will fix shortly

jeremypoulter, 2021-11-07:
Should be fixed now

andriuskr, 2021-11-08:
Thanks, now it is working
[image]

andriuskr, 2021-11-08:
By the way, "Time to full charge" is not very accurate term. I have set to charge battery to 90%. It could fetch this from tesla acc and show not "Time to full charge", but "Time to 90% charge" or "Time to 80% charge" ?

jeremypoulter, 2021-11-08:
Glad to hear that you have got it working. Will keep this open to fix the issue with the login. Can you raise the Time to full charge as a new issue?

ve6rah, 2022-08-13:
Something is still causing me issues here. On both 4.1.2 and 4.1.4 I enter the tokens manually from the "Tesla Tokens" app (same app I use successfully for my home assistant integration) and all I ever see is "Fetching Vehicle Info" and it never gets any farther.

closed 2023-03-31 by glynhudson