OpenEVSE/openevse_esp32_firmware #235: Eco PV Divert doesn't work

cris88, 2021-08-22:
Hello,

I have a problem with PV Divert. Basically it does not regulate as it would be supposed. It is always charged at the maximum amperage set. Grid import and export values ​​are correctly shown on the portal. There are some attached print screens. Thanks.
[image]
[image]

glynhudson, 2021-08-24:
Your config looks correct, please ensure your Grid IE power readings are positive import and negative export. It's also important that the Grid IE feed includes the power consumed by the EVSE. The Grid IE emonPi CT sensor should be clipped around the incoming cable right as it enters the property before any loads. 

Please update to the latest WiFi FW version V4.1.0: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/4.1.0

cris88, 2021-08-24:
hello @glynhudson. Yes Grid IE is in watts, and positive is the value that im importing from grid, when negative the value that im exporting to the grid. Ok i will do the update and give feedback

doppiaemme, 2021-09-08:
> hello @glynhudson. Yes Grid IE is in watts, and positive is the value that im importing from grid, when negative the value that im exporting to the grid. Ok i will do the update and give feedback

I had the VERY same issues with 4.0 which seem disappeared (for now) thankfully to the upgrade to 4.1.0. 
the latter seems so much more stable

closed 2021-09-08 by glynhudson