OpenEVSE/openevse_esp32_firmware #239: Configure hardware button on OpenEVSE

Fortunefavours, 2021-08-29:
We have an OpenEVSE and most of the time we run this in solar divert mode, to mop up any excess.

But every now and then the car needs a proper charge and I want to make it as easy as possible for the whole family to do this.

Is it possible to configure the button on the side of the device so one press will change it into a “full charge just this time” mode, then it’ll default back to PVDivert for future charges?

Mentioned this on OpenEnergyMonitor and @jeremypoulter was kind enough to mention this was already being discussed. A boost mode would be great, especially if it would work without messing with the normal PVDivert/times charging setup.

thanks very much

glynhudson, 2021-08-31:
I agree, this would be a useful feature. However, implementation is tricky since the push button is already used for manual override to stop/start a charging session.

As a first instance, we would need to make it more clear on the LCD when Eco mode is enabled. 

Possibly we could have a config on the web UI to select the function of the button, possible options could be 

1. Manual override (stop/start)
2. Eco Mode (enable /disable)
3. Button disabled

What do you think @jeremypoulter @chris1howell ?

Fortunefavours, 2021-08-31:
Thanks @glynhudson, a config option in the UI makes sense and gives maximum flexibility for different situations. In addition to the three options above, could I request a one-time boost mode, allowing a single full charge before the device returns to its previous (eco mode) settings? This would save having to turn eco-mode back on once you’re done with a full charge. 

If it were possible to read battery charge levels you could have single/double press options configurable, ie:

single press: one time boost mode to 80% then default back to eco-mode.
double press: one time boost mode to 100% then default back to eco-mode

Thanks very much

jeremypoulter, 2021-08-31:
I think there is probably going to be a lot of context sensitive and configurable behavior. IMHO I think we should do something like:

- if sleeping, or in charging with current limit set (eg eco mode), then select from a number of boost options, if we have SoC 80%, 100%, off, else time 30m, 1h, 1h 30m, off.
- else toggle manual override

The boost modes should be fairly simple to do when the generic support for limits is added, which will b part of #225

glynhudson, 2021-09-02:
>if sleeping, or in charging with current limit set (eg eco mode), then select from a number of boost options, if we have SoC 80%, 100%, off, else time 30m, 1h, 1h 30m, off.

I agree, that sounds good. The simplest option for an initial implementation would be just to turn off EcoMode on button press if EcoMode is active. 

When charging in Normal mode, a button press would perform manual override, i.e stop the charge.

We will obviously need to make it clear with a message on the LCD what effect a button press had.

fhteagle, 2021-10-29:
What about two different actions for short press vs long press and hold? Short press is always manual override, long press is a one time charge now at max rate to 100% ?

jeremypoulter, 2021-11-13:
Can be implemented at the same time as #112

jeremypoulter, 2021-11-13:
And #190

thegit, 2022-05-26:
I would like slightly different functionality. I would like to cycle through all; or toggle between preset amperages. I.e. 13amp 16amp 32amp.

TommySharpNZ, 2025-01-30:
Any updates on physical button option? Had no wifi today and couldn’t charge my car…

glynhudson, 2025-01-31:
> Any updates on physical button option? Had no wifi today and couldn’t charge my car…

The default function for the button in current FW is to enable manual override, which will enable you to charge your car even if you can't access the web-portal. This is already present and working in current FW. 

This issue is related to customisable configuration of the physical button.

closed 2026-09-21 by github-actions[bot]