OpenEVSE/openevse_esp32_firmware #246: Home Assistant intergration

jeremypoulter, 2021-09-10:
The current [Home Assistant integration](https://www.home-assistant.io/integrations/openevse/) is a bit outdated and with the new features in both OpenEVSE and Home assistant we can do better:

- Auto discovery
- Setup via Home Assistant UI
- Real time updates via web socket
- Support energy API
- Use newer OpenEVSE APIs

## Sensors
- Current
- Voltage
- Power?
- Session Energy
- Total Energy
- Status
- Temps
- Vehicle connected
- Rssi

## Actions
- Max current
- Charge current
- Manual Override
- Schedule
- Divert mode

https://developers.home-assistant.io/docs/creating_component_index

andriuskr, 2021-09-11:
Is there any ETA for updating integration?  Trying to accomplish charging current regulation in home assistant by whole home usage (limited home electricity input) for a while now.

jeremypoulter, 2021-09-12:
Probably not going to be till next month, but no promises...

doppiaemme, 2021-09-13:
check this out for the time being:

https://gist.github.com/TonyApuzzo/c0fe22458d4dae0effb47b951b464484#file-openevse2-yaml

jeremypoulter, 2021-10-09:
https://github.com/tmjo/charger-card

firstof9, 2021-10-31:
I've been working on this a bit myself:
https://github.com/firstof9/openevse

omriasta, 2021-11-10:
> I've been working on this a bit myself: https://github.com/firstof9/openevse

This integration is great! Exactly what I was looking for. It would be great if you could add an option to use mqtt in the config flow as well!

firstof9, 2021-11-10:
> It would be great if you could add an option to use mqtt in the config flow as well!

MQTT has it's own integration you can use.

This is specifically geared towards using the HTTP API as much as possible.

omriasta, 2021-11-10:
> > It would be great if you could add an option to use mqtt in the config flow as well!
> 
> MQTT has it's own integration you can use.
> 
> This is specifically geared towards using the HTTP API as much as possible.

Yep, I have MQTT setup as well. Was thinking of having them combined all in one integration where one could select if they setup using http or mqtt. My specific setup is I have 2 units, one is on the local network and the other is remote but publishing to my mqtt broker.... I guess big picture if we want to replace the current integration, having both options combined into one would be great for all uses.

firstof9, 2021-11-10:
Please open a feature request, I'll see what I can do, I can't test mqtt right now due to openevse not connecting to my broker for some unknown reason.

jeremypoulter, 2021-11-13:
@firstof9 are you using a secure connection for MQTT? could be certificates is so, but do need to do work on reporting failures. Maybe open another issue and we can discuss there?

drobtravels, 2021-12-10:
Just putting in my 2 cents since I had originally said I would work on this.  Personally I won't be able to use a Home Assistant integration with the new API until #272 is resolved.

For now I'm using the gist [referenced above](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/246#issuecomment-918271516).  Unfortunately its not reliable due to #135

jeremypoulter, 2021-12-10:
Trying to get to both of those, hopefully soon

ruimarinho, 2021-12-22:
@jeremypoulter have you considered using [ESPHome](https://esphome.io/)? It would probably mean a big rewrite but it brings so much out of the box that I thought I'd leave the suggestion here.

firstof9, 2021-12-22:
ESPHome wouldn't really fit for this kind of hardware/project.

jcsogo, 2021-12-28:
I have started to use the integration from @firstof9 (thanks) - I see that there are still some stuff to be implemented, like the ability to set the charge mode or even the manual override. But there is one thing that surprises me, which is that the only way to get the  Level min and max current is through a RAPI command. This information should be provided with the level, as it is the basis to show a proper menu to manually select the current. Right now @firstof9 is hardcoding values up to 48A, which is over what a regular charger would support in Level 2 configuration in Europe. 

@jeremypoulter any plan on providing this information through the API?

Many thanks to you all for your fantastic software.

jeremypoulter, 2021-12-28:
I am currently working on that as part of #140

firstof9, 2021-12-28:
> I see that there are still some stuff to be implemented, like the ability to set the charge mode or even the manual override.

It's still a work in progress, as more API stuff get's fleshed out the move function I can give the integration.

rbray89, 2022-03-07:
Fully MQTT doesn't seem possible at the moment due to RAPI deprivation and no MQTT equivalent for the HTTP API.

glynhudson, 2022-08-04:
> Fully MQTT doesn't seem possible at the moment due to RAPI deprivation and no MQTT equivalent for the HTTP API.

MQTT API is now available since this PR has been merged which should be equivalent to HTTP API https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/379. Big thanks to @KipK 

Se docs: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/docs/mqtt.md

glynhudson, 2022-08-04:
> I've been working on this a bit myself: https://github.com/firstof9/openevse

Nice work! That's a really nice integration. The only thing that's missing is the ability to set a schedule and the ability to manually start / stop a charge.

firstof9, 2022-08-04:
> set a schedule

This will likely be setup as a service call in Home Assistant.

> the ability to manually start / stop a charge.

Are you referring to the `override` on that? If so, this will likely be a service call in Home Assistant as well.

KipK, 2022-08-04:
I've just added a PR adding /schedule /schedule/set /schedule/clear 

There should have everything needed to completely control OpenEvse from Mqtt without any need of RAPI. 
If you see something missing on some use case , please share

jeremypoulter, 2022-08-05:
Home Assistant really should be using the HTTP API, you should probably only use MQTT for control when you can not use HTTP as MQTT really isn't designed for one&to-one comms (it is much better at broadcasting one-to-many events) and although you can do it you will get much more reliable results when using HTTP particularly around error handling.

KipK, 2022-08-05:
Not completely agree but it's a matter of taste. 
With QoS and Last Man Standing it's quiet resilient. 
With many different devices/sensors/actuators using different protocols ( zwave, ZigBee, http, some binary proprietary stuff,etc ), I kinda like to unify everything over MQTT. 
You make me remind I've wanted to add an error topic too 😊

I don't use home-assistant here, but an equivalent solution ( Jeedom ).
With many different devices without Mqtt, it would need different addons for each protocol here.

firstof9, 2022-08-05:
There's a 3rd feature I'm still working on as well for Home Assistant, auto discovery, but I can't seem to find any mDNS for OpenEVSE.

omriasta, 2022-08-05:
It appears this was added at some point...not sure if this helps?
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/260

firstof9, 2022-08-05:
Doesn't seem to work?
```
firstof9@firstof9:~$ avahi-browse -t -r _openevse._tcp
firstof9@firstof9:~$ 
```

omriasta, 2022-08-05:
Hmmm...I ran:
```
avahi-browse -at
avahi-browse -t -r _openevse._tcp
```

And got a response....

firstof9, 2022-08-05:
Ah just showed up after upgrading to `4.1.4`
Strangely after firmware upgrading this is the only time I can get MQTT to connect as well. Weird stuff.

fhteagle, 2023-01-19:
+1 for the integration from @firstof9 . Just installed in my HomeAssistant docker instance, starting to test it out now.

fhteagle, 2023-03-30:
@glynhudson , you can probably close this issue as well. The integration at https://github.com/firstof9/openevse is working well, and has a responsive developer.

glynhudson, 2023-03-31:
Agree, that component works great. I use it myself, @firstof9 is doing a great job 👍

closed 2023-03-31 by glynhudson