OpenEVSE/openevse_esp32_firmware #249: Unable to save MQTT settings

a-u-s-t-i-n, 2021-09-18:
First, I love that this project exists. Thank you to those who work on it.

Second, this might not be an issue with this project, but with my setup. But it might be a bug in openevse-wifi. At the very least it is a request for a more informative error message.

The issue is that I am unable to save MQTT settings, getting a dialog with "Failed to save config" whenever I click the "save" button in the MQTT block. That is, regardless what I do, I am unable to alter the MQTT settings. If I ensure there is an MQTT server at the ip/port currently loaded in my device (put there by me when it was using openevse-wifi 3.0.2-dev), it connects to that server fine. But there seems to be no way to change the MQTT parameters through the web UI for me.

Hardware version: 5.0.1, with a hazzah esp32 (circa 2019). I'd prefer not to have to figure out how to upgrade the hardware's firmware. It also seems that the hardware's firmware version is not the issue here, since I experience the error with both version 3.3.2 and version 4.1.0 of openevse-wifi.

My device was shipped with openevse-wifi version 3.0.2-dev, which had an issue causing it to disconnect from the wifi with some regularity. I finally sat down to figure that out today. First I upgraded to openevse-wifi version 3.3.2 -- the changelogs for some earlier 3.X version suggested that the wifi issue had been fixed. With that version, when I tried to save the MQTT configuration, I get "Failed to save config". I then installed version openevse-wifi version 4.1.0 which has the same issue.

(I'm thinking about reloading an older firmware just to change the MQTT settings, which probably isn't how you'd want users to handle things :).

Could it be my device is out of space? I would be more than happy if this issue were fixed by making the error message more informative.

Thank you.

jeremypoulter, 2021-09-18:
Is it just MQTT setting, or have you tried to save other settings?

Can you open the browser DevTools and see is there are any errors reported on the console?

a-u-s-t-i-n, 2021-09-18:
First: good suggestion. The browser devtools showed I was getting blocked because of cross origin requests. I was using "http://openevse.local" to connect, when I instead used "http://192.168.0.xxx" (the right IP) to connect, the errors went away. It seems that the "save" action hit a URL with the IP in it, which looks like a cross origin request when I'm connected to the .local URL (even though it isn't really -- maybe this is a browser bug?).

Thanks!

Second, it was the other settings on the "services" tab. The settings on the other tabs didn't seem to be affected (though I didn't check them all).

So, I don't know what you can possibly do here. You have no way to tell if I'm connected via "openevse.local" or a specific IP address.  Maybe the change requests can be relative URLs?

I'm going to close the ticket, though. I'm happy with just connecting with the IP when I want to change things.

closed 2021-09-18 by a-u-s-t-i-n