OpenEVSE/openevse_esp32_firmware #261: ESP32 V4.1.0 resets too soon with loss of comm to OpenEVSE

fhteagle, 2021-10-29:
I am working with Chris on troubleshooting wifi on my OpenEVSE v5 (v7.1.3 firmware) + ESP32 (v4.1.0). Current theory is the writing harness between the two is bad, and another harness is enroute to me. In the meantime, wanting to troubleshoot absolutely everything has brought up a possible bug in the wifi firmware.

With V4.1.0 on ESP-32, the wifi AP standalone mode disappears very quickly, every 25 seconds or so when there is no communication with the OpenEVSE. Even if "Wifi Standalone mode" UI button is pushed, or an HTTP firmware push is started, the ESP32 self resets and associated STA connected to standalone AP is dropped, and has to manually re-auth again. This means that a subsequent HTTP firmware flash has no chance of completing.

Is there any way to stop the automatic self reset cycle so a firmware flash can complete?

If not, I suggest this gets added, that either the Wifi standalone mode button or start of HTTP firmware push cancel the self reset until next loss of power.

Thanks!

jeremypoulter, 2021-10-29:
In theory the auto reboot should stop as soon as you connect to the AP, but not checked it in a while...

fhteagle, 2021-10-29:
Not what mine is doing currently. 

Is there any way to know if it's a bad flash causing my symptoms?

jeremypoulter, 2021-10-29:
I wouldn't have thought it is bad flash, but if you check the LED, should block slowly in AP mode, but faster with a client connected,

fhteagle, 2021-10-30:
I hadn't noticed until you mentioned it, but no the LED is not on at all since I flashed v4.1.0.

Time for more troubleshooting......

chris1howell, 2021-10-30:
What version of ESP32 do you have? Did you flash the correct version that
matches your hardware?

fhteagle, 2021-10-31:
I downloaded and unzipped this file to flash to 4.1.0

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/4.1.0/openevse_wifi_v1.bin.zip

glynhudson, 2021-10-31:
Maybe check the voltage on the OpenEVSE ViFi V1 board, your issue sounds like it could be a power supply issue?

fhteagle, 2021-11-12:
Latest data:

1. Verified 12.02VDC (no visible variation in no load voltage on my multimeter) coming from openEVSE power output terminals
2. No change in behavior with replacement openEVSE to wifi harness sent from OpenEVSE store.

Still no wifi information pushed to the front display on power up. 

Still unable to connect to the standalone wifi long enough to attempt a re-flash.

Anything else I can try?

fhteagle, 2021-11-13:
Further update: my wifi board is now being sent back to OpenEVSE Store for a re-flash. Chris' theory is that the wrong firmware got loaded into the card, preventing communication with the EVSE main board. 

I am not sure this bug report should just be closed. Rather, I believe it should be treated as an enhancement request. The enhancement would be to increase the delay before a self-restart, to give a enough time for an over wifi firmware flash (> 5 minutes?), even if comm is lost with the EVSE main board. Or maybe this self restart delay could be triggered by pressing the wifi standalone button in the UI.

And additional enhancement request is for "wrong firmware loading attempted" or "wrong firmware loaded detection or protection logic" was available. No idea how this could be accomplished, exactly.

Thanks for the help and support so far everyone.

jeremypoulter, 2021-11-13:
Hum.... As above, should already be the case, so more this is a bug fox no make sure it doesn't reboot while connected.... That being said the ESP32 has an issue where in gets the data too fast and triggers the reboot

closed 2021-11-13 by glynhudson

fhteagle, 2022-03-11:
Follow up: Chris triaged the unit in question and referred to it as "toast". Replacement hardware is now up and running. Agree issue should be closed until / unless it happens to anyone else.