OpenEVSE/openevse_esp32_firmware #262: Add workflow to upload release assets

jeremypoulter, 2021-10-30:
To make the release process more automated and robust we should use the assets built by the GutHub action for the release.

Option 1)

- Manually create a release in GitHub
- Workflow grabs the associated assets from the build workflow https://github.com/marketplace/actions/download-workflow-artifact

Option 2) 

- On push to `master`
  - create a pre-release if none exists
  - upload assets to release
- Remove pre-release status when ready for release

glynhudson, 2021-10-30:
Either would be fine with me. I don't see it as a particuarly big problem to manually upload the releases manually, but I guess it would make the process more robust and eliminate user error.

jeremypoulter, 2021-10-31:
This is mainly to help with certification and making sure the release process is consistent.

jeremypoulter, 2021-11-03:
I have been experimenting with option 2) https://github.com/jeremypoulter/ESP32_WiFi_V4.x/releases looks to work well. After each commit to my test branch the 'latest' pre-release is updated, the .bin can be downloaded/tested, then when ready the latest release can be edited and updated with the release number, tag, release notes and remove the pre-release setting. On the next merge to master a new dev pre-release is created.

@glynhudson / @chris1howell what do you think? 

Also do we need the bootloader.bin and partitions.bin on each release?

glynhudson, 2021-11-03:
Nice, that looks great. Where are the older releases, I can't see them listed? 

>Also do we need the bootloader.bin and partitions.bin on each release?

No, that's not essential

jeremypoulter, 2021-11-03:
Cool, I will create a PR

commit d22c016b (2021-11-05): Automated Release process

Extended workflow to create/update a pre-release called latest after each commit/merge to master, uploading the generated assets to that release. The binaries can then be downloaded and tested and the release promoted to a full release if as needed.

Fixes #262
PR #266 (2021-11-07): Automated Release process
Extended workflow to create/update a pre-release called latest after each commit/merge to master, uploading the generated assets to that release. The binaries can then be downloaded and tested and the release promoted to a full release if as needed.

Fixes #262
closed 2021-11-07 by glynhudson