OpenEVSE/openevse_esp32_firmware #263: [BUG] Websocket data does not match documentation

firstof9, 2021-11-03:
In the documentation `wattsec` and `watthour` should update via the HTTP websocket, these fields are missing from the websocket updates.

jeremypoulter, 2021-11-10:
Yeah I am updating the docs currently for v4 (#9) but the `wattsec` and `watthour` are terrible names so I have added `session_energy` and `total_energy` and that is both in the web socket, `/status` and MQTT:

[image]

[image]

[image]

firstof9, 2021-11-10:
I assume these will be in the next firmware update then?

jeremypoulter, 2021-11-10:
If not in 4.1 then yeah, you can try the [pre-release](https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/tag/latest)

ruimarinho, 2021-12-23:
Confirmed on 4.1.1.dev.

[image]

closed 2023-03-31 by glynhudson