OpenEVSE/openevse_esp32_firmware #264: Display session energy as kW with one decimal place

glynhudson, 2021-11-04:
Currently, the session energy is displayed in Wh with two decimal places! The results in a rather large number, and when the session energy > 10kWh the LCD display is truncated and the 'h' symbol is missing, which results in confusion since it then looks like a power value rather than energy.

I propose display session energy as kW with one decimal place, what do you think @chris1howell ?

[image]

chris1howell, 2021-11-04:
Yep, sounds good.

jeremypoulter, 2021-11-04:
Been meaning to make this a bit more sane. I would propose a degree of scaling based on the size, ie Wh for > 4000, kWh > 4,000,000, etc and that would just be a generic thing across the board, ie total energy, current power, etc

jeremypoulter, 2021-11-04:
Or given the LCD has a specific number of characters maybe just base on that and give the max precision for the available space

closed 2021-11-29 by glynhudson

glynhudson, 2021-12-01:
Nice work, this has fixed the issue perfectly :+1: 

[image]