OpenEVSE/openevse_esp32_firmware #268: Time to full (set) charge

andriuskr, 2021-11-10:
[image]
"Time to full charge" is not very accurate term if vehicle is set to charge battery to 90%. It could fetch this from tesla acc and show not "Time to full charge", but "Time to 90% charge" or "Time to 80% charge"

glynhudson, 2021-11-13:
@jeremypoulter is the 'time to charge' metric already fetched from the Tesla API? This metric is currently not used for any purpose, I vote removing it to avoid confusing at the moment.

jeremypoulter, 2021-11-13:
Initially we can probably just change it to 'time to complete' or something more generic, think it is just for information, but probably quite useful, especially on the LCD.

closed 2023-03-31 by glynhudson