OpenEVSE/openevse_esp32_firmware #271: [BUG] MQTT connection not connecting

firstof9, 2021-11-13:
OpenEVSE Wifi Firmware: 4.1.0
OpenEVSE Firmware: 7.1.6

When enabling MQTT the only status I get is `Connected: No`.

I left the settings as they were from Wifi FW 4.0.0, after the firmware update I am no longer able to get connected to my MQTT broker.
It is not a secure connection.

[image]

I use the same settings in my MQTT-Exploer and it connects just fine:
[image]

cdriget, 2022-06-28:
Hello,
Any update on this particular bug ?
I am facing exactly the same issue.

firstof9, 2022-06-28:
Odd workaround, but I've done this and over time the connection would go MIA.

glynhudson, 2023-03-31:
Is this still an issue with latest FW version? I'm not able to replicate

firstof9, 2023-03-31:
> Is this still an issue with latest FW version?

In 4.1.7 my MQTT connection goes away after a while still, upon restarting wifi it comes back for a bit, then eventually goes away.

ericvogel84, 2023-06-08:
I'm in 4.2.2, and I can't get it to connect at all. Debug is just showing {"mqtt_connected":0} with no errors. 

I too am using the exact settings as I have in MQTT-Explorer, and I have tried rebooting wifi - nothing. It never connects. 

Does anyone have any tips?

closed 2026-09-21 by github-actions[bot]