OpenEVSE/openevse_esp32_firmware #272: Clarification on Manual Override energy_limit and time_limit

drobtravels, 2021-11-14:
With the HTTP API, I see manual override supports [energy_limit and time_limit properties](https://openevse.stoplight.io/docs/openevse-wifi-v4/b3A6MjA2MTYyMzU-set-the-manual-override)

I did a few experiments and am unclear how to use this.

1. What unit is `energy_limit` in?
2. What unit is `time_limit` in?
3. Can you set both together?  It doesn't error if you do this, but I'm unclear on the expected behavior
4. If a session is already in progress, does this limit apply from _now_ or from _session start_
5. I was planning on using this Manual Override API from Home Assistant as the main way to start charging.  IE  I want Home Assistant when to start charging, and for how long.  Is that the intention, or is there a different option I should use?

Thanks!

drobtravels, 2021-11-14:
Oh also, what's the difference between `charge_current` and `max_current`?

jeremypoulter, 2021-11-14:
> 1. What unit is `energy_limit` in?
> 2. What unit is `time_limit` in?

Unfortunately, these are not implemented yet, but probably will be seconds. Will probably implement at the same time as #225 

> 3. Can you set both together? It doesn't error if you do this, but I'm unclear on the expected behavior

TBD, but probably yes and the first limit reached will be the winner and stop the charge

> 4. If a session is already in progress, does this limit apply from now or from session start

Will be from the start of the session

> 5. I was planning on using this Manual Override API from Home Assistant as the main way to start charging. IE I want Home Assistant when to start charging, and for how long. Is that the intention, or is there a different option I should use?

It kind of depends on your use case, but I would recommend the [/claims](https://openevse.stoplight.io/docs/openevse-wifi-v4/b3A6MjA2MTYyNDA-make-update-an-evse-claim) endpoint, essentially the same, but runs at a lower priority so you can still use the manual override from the GUI/Button to override the Home Assistant behaviour

> Oh also, what's the difference between `charge_current` and `max_current`?

I am not totally sure that both needed, but `charge_current` is generally what you should use, `max_current` is there to dynamically alter the max current  in load sharing situations while still allowing things like PV divert to work so the charge current is the minimum of `charge_current` <= `max_current` <= configured current (TBD) <= hard coded limit

drobtravels, 2021-11-14:
Thanks for the detailed reply!

> Unfortunately, these are not implemented yet, but probably will be seconds. 

Ah, that explains the strange results I'm seeing.  I thought I had the wrong units.  I assume this is also not yet implemented for `/claims`? So right now charge with time limit can only be done via RAPI?

jeremypoulter, 2021-11-14:
Correct, these can only be set via RAPI, but be warned this will break in a later 4.x release

closed 2023-03-31 by glynhudson