OpenEVSE/openevse_esp32_firmware #273: Default NTP server name

jeremypoulter, 2021-11-17:
We shouldn't really be using  `pool.ntp.org` as a default NTP server (see <https://www.ntppool.org/en/vendors.html>), we should [apply](https://manage.ntppool.org/manage/vendor) for something like `openevse.pool.ntp.org` and set that as the default.

jeremypoulter, 2021-11-17:
@chris1howell would you be able to look at setting up this?

ausil, 2022-02-10:
+1 for this

RAR, 2026-09-21:
The DHCP part of this thread is implemented in #1289 (firmware: prefer the NTP server from DHCP option 42, fall back to the configured host if it stops answering, `sntp_dhcp` toggle defaulting on) with the UI in OpenEVSE/openevse-gui-nightshift#143. It needs a test on a network whose DHCP server hands out option 42 — checklist in the PR. The original ask here (an `openevse.pool.ntp.org` vendor zone as the default) is an admin step at manage.ntppool.org plus a one-line change to `SNTP_DEFAULT_HOST`, still open.

RAR, 2026-09-21:
And for the original ask: #1290 switches the fresh-install default to `time.cloudflare.com` (free for any client, no vendor registration to maintain) rather than applying for a pool.ntp.org vendor zone; GUI placeholders in OpenEVSE/openevse-gui-nightshift#144. Existing installs keep their stored `sntp_hostname`.

jeremypoulter, 2026-09-22:
Just to mirror my comment on the PR this ticket was about doing the registration, rather than switching to another default.

RAR, 2026-09-22:
Totally - but after 5 years that hasn't happened, we're still using pool.ntp.org which they ask us not to.   We can leave it and close this out - the NTP via DHCP ask is under https://github.com/OpenEVSE/openevse_esp32_firmware/pull/1289.  

Just don't want this open 5 years from now collecting dust.

jeremypoulter, 2026-09-22:
Agreed @chris1howell is this something you can look at? Either register with ntp.org or accept the PR?

RAR, 2026-09-22:
> It would still be good to take the dhcp provided NTP server!

[image]

will be in next release.

closed 2026-09-22 by chris1howell