OpenEVSE/openevse_esp32_firmware #281: Set Current - Add option for volatile temporary setting

chris1howell, 2021-12-01:
Received Feature request via support.

Add GUI option for temporary current setting that is not saved to memory by default.

Use case, a portable station used on 120v 16A and occasionally TT30 24A. Allow GUI option that would send $SC 24 V so when the station is powered down it will revert back to 16A and not overload the circuit if user forgets to set the station back.

Save to memory slider?

jeremypoulter, 2021-12-01:
Kind of related to #140, will consider how best to do this, hadn't considered that you would temporarily raise the current so a lot of design choices have been made to prevent this as a safety measure... Will have to check that this is how the OpenEVSE firmware allows this, but possibly could allow making a claim with  `max_charge_current` > the configured soft limit (but always less than the hardware limit)

rbray89, 2021-12-28:
Yeah, I have a similar need with two stations sharing the same breaker. I want to use MQTT to provide a slider to shift current percentage between stations as I prioritize one vehicle over the other.

glynhudson, 2023-03-31:
I believe this is now implemented in V2 UI? @KipK ?

KipK, 2023-03-31:
yep, charge current main slider is a manual override claim. Everything from the main screen are just dedicated to current session and not saved in flash ( but  Eco / Boost switch )
Talking about the last one, do you think we should keep this switch writable ? or should I move from the /config {"charge_mode"} prop to the volatile equivalent ( /divertmode ) endpoint ?

sspearing, 2023-09-30:
From what I'm seeing GUI V2 does not yet have the option to temporarily increase charge current above the default non-volatile current setting.  The RAPI command "$SC 24 V" handled this perfectly by putting this higher setting into volatile memory.  Once the unit was unplugged, it would default the L1 back to 12A.
To expand on our use case:  We commonly take our car to a campground with L1 30A plugins (24A net).  We can also charge from standard duplex outlets L1 15A (12A net).  For safety, I appreciated that the OpenEVSE would default back to L1 12A so we couldn't overload a duplex outlet.

closed 2026-09-21 by github-actions[bot]