OpenEVSE/openevse_esp32_firmware #282: Divert Mode Issues

pbix, 2021-12-04:
Struggling with divert mode issues and have some questions.  

1) What exactly is the "pilot" value displayed in the UI?  Where does this come from?  I would have assumed that this is the maximum charge rate being communicated to the vehicle but that does not seem to be the case.  See below where the pilot is showing "12" and the vehicle is charging at 40A.

[image]

2) How exactly does one set the maximum current the charger can support?  I am wired for 40A but it seems that Setup ->Max Current and the Current->Level 2 maximum current have no relationship to what happens with the charger.  See below where all three are in contradiction.

[image]

Finally, I have observed that a reset of the Wifi module and snap the charger out of the mode where the pilot current gets stuck.  Its all leading to a very confusing experience.

I have a Tesla and I am updating the MQTT reference every 10 seconds.

pbix, 2021-12-16:
I can add to this post that if I reset the Wifi module I get out of this mode, at least until I again toggle the Charge Option Normal/Divert switch.  It seems that changing this toggle and/or playing around with the Max Current setting will get the module in a state which pretty much ignores the divert power reference.

ruimarinho, 2021-12-22:
I'm also confused on the pilot value. I thought that it was the max current possibly delivered to the car, doesn't look like it:

[image]

danpadure, 2022-02-05:
I noticed the same issue.

https://youtu.be/SLNhvLQx0bo

jeremypoulter, 2022-09-24:
This was probably fixed by #329 can this be retested in the latest firmware?

pbix, 2022-09-24:
The commit you reference is in 4.1.3 which I have been using for some time.  I cannot say that I have seen this exact issue since updating but I am struggling with the related issue that the divert mode does not follow the divert reference which is the subject of #387 that I know you are working on.

I am going to close this one and watch what happens over there.  

Thanks for your efforts on this.

closed 2022-09-24 by pbix