OpenEVSE/openevse_esp32_firmware #283: Divert Mode - Charge without available - Beta version

ayasystems, 2021-12-05:
The car was not plugged in, once it was plugged in it started charging, but there was not enough solar energy.

This happened with version 4.1.0 and also happens with the latest beta. 

If I deactivate divert-mode and activate it again, it no longer starts charging because there is not enough solar power (this is correct).

[image]

openevse_huzzah32 beta is showed as 4.1.0 version, but as you can see it is the beta version

[image]

ayasystems, 2021-12-11:
[image]

I think the the problem is that eco mode is not module charge current. It is not working

ruimarinho, 2021-12-22:
I'm on latest dev version and experiencing the same issue:

[image]

Vehicle was supposed to wait for power from solar but started charging right away. Notice the confusing wording "charging from solar | Charge rate: 0A"

tschiex, 2022-02-05:
Same here. And as soon as I switch the "Normal (fast) / Eco (PV divert)" button to "fast" and then back to " eco", charging stops, as it should.

tschiex, 2022-02-05:
BTW, this looks like a duplicate of #279 .

ayasystems, 2022-03-24:
The same error with the last beta 

[image]

As yoy can see without manual_override and without solar charger is in state 1

[image]

After reboot esp it is fine....

[image]

I

closed 2022-05-25 by glynhudson

ayasystems, 2022-07-08:
@glynhudson The issue was closed but it was not fixed