OpenEVSE/openevse_esp32_firmware #288: Vehicle State (SoC, Range, ETA) via MQTT only work if Divert Enabled

spafrost, 2021-12-09:
If MQTT is configured as the source for data in the "Vehicle" tab, it does not populate unless "Enable Solar PV Divert" is enabled in the "Services" tab. 

It looks like this is down to code from #67 , which only subscribes to the configured topics is Divert is enabled. 

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/38f2fd83b719e2ec60da2f526b6756a8dca7a546/src/mqtt.cpp#L208-L225

Had been pulling my hair out trying to figure out what I had misconfigured (I don't have PV, so not enabled)! Not sure if this is intentional or a bug. 

Ps. First Git interaction for a long time, shout if I have missed anything

adityathakekar, 2021-12-14:
Thanks for sharing. Great observation. Trying to find the bug.

billatq, 2022-02-13:
I was thinking about setting this up so that I can get the SoC from my Fiat 500e, since I don't have telematics for it. It looks like this is as simple as just moving those vehicle subscribes out, since the event handler higher up in the code should do the right thing when it gets an event.

Is this just a matter of making that change and testing it? I'm happy to give that a go if nobody else is working on it.

jeremypoulter, 2022-02-13:
That would be great, basically fork the repository, make the changes (probably on a branch), test and submit a Pull Request.

billatq, 2022-02-13:
Got my build environment set up, moved the vehicle charging specifics out of the `if` block, did an OTA update, and then tested sending messages over MQTT. Verified that the web UI remains hidden until it sees a subscribed message, and then updates. Solar PV divert is not enabled.

[image]

closed 2022-02-14 by jeremypoulter