OpenEVSE/openevse_esp32_firmware #289: Mennekes lock

glynhudson, 2021-12-13:
The updated Mennekes lock in controller V8.1.0 https://github.com/lincomatic/open_evse/issues/141 has both an automatic and a manual mode. In manual mode, the lock can be controlled via RAPI. 

A useful basic function for this manual mode would be to provide a toggle on the web interface to set the lock mode:

1. Automatic 
2. Permanently Locked 
3. Permanently Unlocked 

Manual mode is non-volatile, therefore must be enabled on boot. It would be a good idea to show the lock status on the `/status` info page. 

```
S5 A|M|0|1 - Mennekes lock setting
   A = enable automatic mode - locked when connected, unlocked otherwise
   M = enable manual control mode
   0 = unlock (valid only in manual mode)
   1 = lock (valid only in manual mode)
   n.b. requires MENNEKES_LOCK. manual mode is volatile - always boots in automatic mode
```

```
G5 - get Mennekes settings
 response: $OK state mode
   state: 0 = unlocked
          1 = locked
   mode: A = automatic mode - locked when connected, unlocked otherwise
         M = manual control mode
   Note: lock mode is also indicated by ECVF_MENNEKES_MANUAL
   n.b. requires MENNEKES_LOCK
```

The lock will also need to be implemented via OCPP and HTTP/MQTT API.

KipK, 2023-04-20:
I'm planning to add the Menneke lock actuator to my EVSE for testing and finilising this feature. 
@jeremypoulter I haven't checked yet, does the openevse_lib already implemented it?

jeremypoulter, 2023-04-20:
I don't think so

closed 2026-09-21 by github-actions[bot]