OpenEVSE/openevse_esp32_firmware #290: webpage no longer working after some days

danpadure, 2021-12-15:
I am constantly having this issue: For some days (5 maybe) openevse wifi works fine
Then for unknown reason(s) the web page no longer works. I can ping it, it even answers to api requests like /config
The issue could be related to alternating access from PC to mobile phone, or could be just a coincidance

Since there are no logs, how could I check what crashes?  
Can I login using ssh or telnet?

jeremypoulter, 2021-12-15:
If it is responding to API calls there is no reason that the UI shouldn't load, all served the same. Can you be more specific as to what doesn't work? Is the browser failing to download the HTML? 

Cheers,
Jeremy

danpadure, 2021-12-15:
if I press refresh it stops at loading please wait 0/17
If I open it with a new browser, nothing loads up, times out after a few min
Is there any way to make it log and then check somehow what's wrong?
or a reboot API call?

I don't know if it's related, but when this happens, diverter no longer works, it charges the car even without solar. I remember that from API I noticed override was disabled, but it was charging still.

jeremypoulter, 2021-12-15:
If you open the Dev tools on the browser you should see an error there as to why it stops loading.  

To reboot you can use the /restart endpoint.

danpadure, 2021-12-15:
Sorry, do you mean DEBUG button or OpenEVSE?
Both open some form or console where I can see informations

/restart works indeed, thanks!

jeremypoulter, 2021-12-15:
Most browsers have built in developer tools, eg Chrome:
[image]

danpadure, 2021-12-15:
oh, ok, I know about devtools in the browser.  
Will I see the error there? ok

jeremypoulter, 2021-12-24:
Should do, either JS errors or network errors, but that is normally a good place to start.

chaseadam, 2022-01-04:
I can report similar behavior, but mine doesn't happen after 5 days. This only occurred after activating MQTT.

The transfer appears to stall. This is what I get from an authenticated cURL request for `/` after several minutes.

```
*   Trying 192.168.5.238:80...
* Connected to 192.168.5.238 (192.168.5.238) port 80 (#0)
> GET / HTTP/1.1
> Host: 192.168.5.238
> Accept-Encoding: deflate, gzip, br
> User-Agent: Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:94.0) Gecko/20100101 Firefox/94.0
> Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,*/*;q=0.8
> Accept-Language: en-US,en;q=0.5
> Authorization: Basic YWNoYXNlbjpJbnNlY3VyZTEh
> Connection: keep-alive
> Upgrade-Insecure-Requests: 1
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: Mongoose/6.14
< Connection: close
< Content-Type: text/html
< Content-Length: 40491
< 
* Recv failure: Connection reset by peer
* Closing connection 0
curl: (56) Recv failure: Connection reset by peer
<!doctype html> <%  
```

jeremypoulter, 2022-01-04:
I did see a similar issue on one of my test boards interesting that it is related to MQTT, there is a lot more going out via MQTT these days so maybe there is a conflict. I will have to check it out.

danpadure, 2022-01-04:
Since I opened this issue, I have not encountered the issue anymore, so maybe this is the fix ;)
I think it could be related to how often you change charging parameters in the webpage.

I also noticed that sometimes the max charging speed gets set at 6A, and sometimes the car is charging even if set to only use PV power...   This gets fixed arter I switch the power to normal and back to diverted again

Not sure if all these issues are related or not

kabutotx, 2022-01-20:
I have been having problems with loading since 4.1.1. Will see just the Loading, please wait (0/18). Refreshing a couple of times will eventually get it to work.  Happens both on my Android phone and Win 10 laptop.  Below is the error in console log I get when it won't load.
`lib.js:1 Uncaught TypeError: e.status is not a function
    at EventLogViewModel.addLog (home.js:1:49177)
    at N.ic.lc (home.js:1:24744)
    at Function.notifySubscribers (lib.js:1:115612)
    at Function.xa (lib.js:1:117935)
    at t (lib.js:1:117674)
    at i (lib.js:1:174139)
    at lib.js:1:173630
    at V (lib.js:1:169693)
    at i (lib.js:1:173319)
    at Object.M.fromJS (lib.js:1:174249)`

dconlon, 2022-02-26:
I have the same behaviour with 4.1.0 (which my unit came with) and 4.1.1 which I have upgraded to.

After a period of days, approximately 1 in 4 HTTP requests fail with the failure rate gradually increasing to the point that the unit needs a power cycle (because /restart is failing).  The unit ceases to POST data to Emoncms too.

* I have MQTT and Emoncms enabled
* I have software POSTing to /claims to vary charge_current during charges

[image]
[image]
[image]

closed 2022-05-25 by glynhudson