OpenEVSE/openevse_esp32_firmware #291: Recommended pinout configuration for ESP32 DevKitC v4 ESP32-WROOM-32U

ruimarinho, 2021-12-17:
Hi @jeremypoulter,

Could you suggest a pinout configuration for platformio.ini for the ESP32 DevKitC v4 ESP32-WROOM-32U board ([Aliexpress](https://www.aliexpress.com/item/4000090521976.html?spm=a2g0s.9042311.0.0.7e214c4dEyDasb))?

I have the official Huzzah32 but I'm giving the IPEX board a try (ESP32-WROOM-32U) to see if the external antenna helps with my constant MQTT dropouts.

Thank you.

jeremypoulter, 2021-12-17:
I think the same pinout as the NodeMCU-32S should work, with the default serial 1 pins connected to the OpenEVSE

ruimarinho, 2021-12-17:
Thanks! Does this board require the configuration of any neopixel stuff?

jeremypoulter, 2021-12-17:
No, no Neopixel stuff for that board

ruimarinho, 2021-12-19:
Looking at the schematics, wouldn't this board be limited to one serial port (TXD0/RXD0), therefore `RAPI_PORT=Serial`? 
[image]

How do I figure out which pin controls the onboard LED? Sorry for the questions :)

jeremypoulter, 2021-12-21:
Assuming it is this board: https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf I think the on-board LED is directly linked to the power so can not be controlled. 

The `TXD0`/`RXD0` are connected to the USB serial so not really suitable for OpenEVSE comms so you need to use one of the other UARTs, these are not shown on the schematic, but if you look at a [pinout](https://www.theengineeringprojects.com/wp-content/uploads/2020/12/introduction-to-esp32-3.png) you will see there are additional UARTs on `GPIO9`/`GPIO10` and `GPIO16`/`GPIO17`, the latter (Serial2) is probably the best as `GPIO9`/`GPIO10` conflict with the internal flash. You can also configure the UARTs to any pins.

ruimarinho, 2021-12-22:
@jeremypoulter got it, thanks! In that case (assuming `-D RAPI_PORT=Serial2`), should/could I do `-D DEBUG_PORT=Serial` to spy on messages? Although it seems like this is the default already.. https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/3e2859fc0b03c728e1b3d936a92152140539e52c/src/debug.cpp#L5

jeremypoulter, 2021-12-22:
Correct, the default is to use `Serial` as the debug.

ruimarinho, 2021-12-22:
Thanks! Everything is working now :) what would be the best way to simulate the `WIFI_BUTTON` on GPIO0? I have EN and Boot buttons, but those are hardwired for ESP32's reset and download function (which I don't really know what it means).

jeremypoulter, 2021-12-23:
The boot button will be connected to `GPIO0` so should be ok to use as the `WIFI_BUTTON`

wrightwells, 2022-01-13:
I’ve connected an ESP32 WROOM 32U for the external antenna and complied using the huzzah32_dev env.
It connects to the EVSE board and all commands work but when powered from the 5v outlet of the EVSE board, the ESVE RGB display goes DIM and pluses. When the WIFI board is not connected or powered seperately the RGB display is very bright and readable?
As the openevse Wi-Fi board uses the 12v pin is this expected?

jeremypoulter, 2022-01-14:
The 5v line on the OpenEVSE module can not supply sufficient power for the ESP32, you have to take the 12v and use a buck converter to drop the voltage down to 5v

wrightwells, 2022-01-14:
Thanks , buck installed and I up and running.

ruimarinho, 2022-01-16:
In my case, I've used the `openevse_nodemcu-32s` environment and it's working great. 

```
pio run  -e openevse_nodemcu-32s -t upload
```

[image]

closed 2022-01-16 by ruimarinho