OpenEVSE/openevse_esp32_firmware #292: Voltage topic for three-phase version

ruimarinho, 2021-12-22:
Hi,

I have the three-phase version and in my case, the voltage is read by a Shelly EM3. I have three different voltage readings for each phase, but the OpenEVSE system only lets me configure one.

Is it possible to take the right voltage reading/topic according to the phase in use or the total when all are in use?

jeremypoulter, 2021-12-24:
@glynhudson correct me if I am wrong, but the OpenEVSE only has a single current sensor so can only measure the current on a single phase. I assume it just multiplies by 3 for 3 phase, so knowing the voltage on all three phases won't really help.

Adminius, 2022-01-28:
I would also say it's not necessary.
Let's say your readings are: L1: 225V, L2: 227 V, L3:229V (I don't think the difference between phases will be much bigger)
with a current of 16A we have 3600W / 3632W / 3664W. I would say it is in tolerance range (because it is only around 2% difference) :)

jeremypoulter, 2022-05-19:
Closing because of the hardware limitation

closed 2022-05-19 by jeremypoulter