OpenEVSE/openevse_esp32_firmware #295: LCD: Bad sequence statuts display

Tifaifai, 2022-01-03:
Hi,
On my OpenEVSE, Configuration : Firmware 7.1.3 + Wifi 4.1.1 (Hardware OpenEVSE Wifi v1).

On the LCD, The second line is ok with informations cycle (energy, etc.)
The first line display bad status sequence.

Sequence:
  Restart Firmware
  Ready - Green Led - No vehicle connected
  Charging - Blue Led - Vehicle connected - Charge
  Connected - yellow Led - Vehicle connected - No Charge
==> Bug:  Connected - Green Led - No vehicle connected

I must restart firmware but identiccaly sequence
  
Thank you for your great job

jeremypoulter, 2022-01-03:
Have been able to replicate this, will take a look.

Tifaifai, 2022-01-05:
Use case sequence #2: 
Restart Firmware
Ready - Green Led - No vehicle connected
Push 'Pause' Charge option
L0: zzz Sleepling zzz / L1: Manage override - Purple Led
Push 'Clear' Charge option
=> Bug: L0: zzz Sleeping zzz - Green Led

Tifaifai, 2022-01-12:
if you generate a 4.1.2 release, I want well to test this. (link to download ?)

jeremypoulter, 2022-01-12:
You can grab the assets from the PR build, https://github.com/OpenEVSE/ESP32_WiFi_V4.x/actions/runs/1670141244

Tifaifai, 2022-01-12:
I just finished testing and the PR #298 is perfect! looking forward to the release 4.2 !

PR #298 (2022-01-21): Latest OpenEVSE library with LCD space fix
Fixes #295
closed 2022-01-12 by Tifaifai