OpenEVSE/openevse_esp32_firmware #302: Timer must be set in UTC

drobtravels, 2022-01-24:
When setting the timer via web GUI it uses UTC time.

This is confusing as the time under "Setup" directly below it shows local time

fhteagle, 2022-03-12:
I'm not seeing this behavior in 4.1.0. Timer clearly aligns with my selected local time zone, which is then updated via NTP....

fhteagle, 2022-03-13:
New clue to the possible root cause of this:

Daylight savings time just happened last night on the US. Time was correctly set before I went to bed using NTP method. I woke up this morning, and the clock had not gone forward an hour, but closer to an hour and a half for some reason. So I started mucking about with manual and browser based time setting. Both of those result in UTC being the time zone set and displayed. 

However, when I went back to NTP mode, the UTC offset for my timezone (UTC -6 at the moment) was applied twice and in the wrong direction. So instead of getting 08:22 local from 14:22 UTC, the EVSE wants to update from NTP to be 20:22 local  on the correct date instead.

I am now wondering if this was also causing the OP's stated issue.

jeremypoulter, 2023-03-11:
Possibly fixed by #569, please cleack

closed 2023-03-31 by glynhudson