OpenEVSE/openevse_esp32_firmware #307: Pause button broken

jeremypoulter, 2022-01-30:
As reported in #296 

> However, the 'pause' button seems to be broken. When I click pause, nothing happens and the error home.js:1 Uncaught TypeError: o.openevse.isPaused is not a function is displayed in the console.
> 
> [image]

closed 2022-01-31 by glynhudson

glynhudson, 2022-02-01:
Mm I seem to still be having the same behaviour testing with the fix PR. I've cleared the cache and tested in private browsing mode. 

[image]

jeremypoulter, 2022-02-01:
Needed the GUI changes merging, forgot to create a PR, should be fixed not

glynhudson, 2022-02-02:
Ok, no worries. I see the GUI PR is now merged: https://github.com/OpenEVSE/openevse_wifi_gui/commit/6c52c6b967b4f9104f7a57838f4385d460b34bff

Should I test again? Will I need to compile locally to pull in the new GUI? 

Is it possible to test a PR without merging using github actions artefacts? Or do PRs need to be merged and compiled locally?

jeremypoulter, 2022-02-03:
The builds are all done as part of the 'Checks', see https://github.com/OpenEVSE/ESP32_WiFi_V4.x/actions/runs/1770122468, I guess it is a little hidden if they pass but look for the green tick against the last commit or the checks section at the bottom.

PR #309 (2022-01-31): Fix for manual override not working
#296 removed the `isPaused` method on `OpenEVSEViewModel` as it ws depricated and replaced by the same method on `StatusViewModel`. Updated the refrence to point to the new location.

Fixes #307
commit f8dc39bc (2022-01-31): Fix for manual override not working

#296 removed the `isPaused` method on `OpenEVSEViewModel` as it ws depricated and replaced by the same method on `StatusViewModel`. Updated the refrence to point to the new location.

Fixes #307