OpenEVSE/openevse_esp32_firmware #311: Problem Synchronisation $SC / web interface / MQTT & Firmware

loraraspi91, 2022-02-03:
When I change the current setting via the Rapi interface (or via mqtt)  for example: "$SC 14 V". 

The current is correctly changed in the firmware (I'm using the version 7.1.3).  But the information on the LCD is not updated also in the webpages. 
If I change via the web interface, everything works fine. It is like there is an issue about the synchronisation when the setting is directly changed by the RAPI interface (MQTT or Serial).

jeremypoulter, 2022-02-03:
Using RAPI to change settings is not supported and will be disabled in future releases. Please use the HTTP API to set the current.

rbray89, 2022-03-07:
What is the proper MQTT topic so set the current?

mrjackson, 2022-03-12:
> What is the proper MQTT topic so set the current?

I've been trying to figure this out for a while and just used RAPI because it works.
I do think I may have figured it out after looking again now, but it acts the same (web doesn't update, ect)
    curl --request POST --url http://openevse.local/override --header 'Content-Type: application/json' --data '{ "charge_current": "7" }'

KipK, 2022-05-19:
I've felt in the same trap, trying to change SC value over MQTT using RAPI.

What's strange is SC value is updated after you've rebooted the device, so it's half working.

I'd prefer to use MQTT over HTTP for controlling it, is there a planned way to ?

KipK, 2022-05-19:
I've played a bit with the EVSE json API.
Same as @mrjackson , I don't have the value updated on LCD, interface and mqtt.
But I feel confused between override and claims functions .
I haven't found any docs about what claim vs override is for?
I'm trying to adjust in realtime my charging current with what is still available on my energy plan. My heaters have priority against charging current. 
Which of booth should I use?
How do you set priority? ( On the API doc there's o priority settings for POST )

jeremypoulter, 2022-05-19:
The MQTT API will be implemented in #358 . WRT `/claim` vs `/override`, the `/claim` API gives low level access to the claims system on top of which all the other 'smart' features are built. The `/override` API is for a specific use case of manually overriding the current status, so in general if you are implementing an automatic behaviour you should probably use the `/claim` API, if you are intending to give direct control to a person then the `/override' API is probably what you want.

closed 2022-05-19 by jeremypoulter

KipK, 2022-05-20:
Thanks for those answers, I get it now. 
I'm exploring the code now, should have done that first :)
I see the mqtt extended part has not been coded yet, perhaps I can help? Have some experiences with Espressif.  I will have some spare time beginning of June.

jeremypoulter, 2022-05-21:
That would be great, any contributions are always welcome. Let me know if you have any questions.

jeremypoulter, 2022-05-21:
> And about the title of the ticket? it seems either using Rapi or the new http API, the values are not updated on LCD nor interface/mqtt

Updating via the HTTP API should be fixed in the latest version as far as I am aware, but updating via RAPI is not supported, note will it be.