OpenEVSE/openevse_esp32_firmware #312: PV divert problems ONLY when PV ratio is less than 1

danpadure, 2022-02-10:
I think I found a reason for strange PV divert behaviour.
So I think these issues only occur when the available diverted PV power is LESS than 6A. And ONLY when you start the charging without enough PV power, so it has to wait for the sun
I think there's something wrong in the code for when power is under 6A

Here is a graph showing the issue 
[image]

Another more detailed graph
GREEN= total PV OUTPUT
BLUE=grid import/export (negative is export)
RED=OpenEVSE Current (A)

[image]

OpenEVSE started charging when export was -300W as it's coifigured to start at 0.2 PV power ratio.
But started at max current (16A).
It operated at that speed until the PV power ratio was 1.
Then after that it operated normally, increasing the current as the panels produced more and more power
I also remember that during the autum when I had configured to start at 1 ratio I had no such issues

pdhoogh, 2022-05-20:
I have seen similar behavior, but it is not predictable. With "_Required PV power ratio_" < 1.0, I have seen the desired behavior when there was more than 6A available first and the PV power dropped below 6A, the current stayed at 6A until minimum ratio was crossed and then charge stopped. Perfect.

But indeed if power is building up from below minimum ratio times 6A and crosses that ratio, the EVSE initiates charge and goes to Max Current.

The way I would think this should work - that is just me - is that when there is less than 6A available, the EVSE would still allow charging at 6A up to the point where the available current is (Required PV power ratio) times 6A. Do I understand this right?

glynhudson, 2023-03-31:
I've also seen this behaviour in recent build 

[image]

danpadure, 2023-03-31:
I think that if I set minimum ratio higher than 0.3, then the problem does not occur

closed 2023-04-05 by glynhudson