OpenEVSE/openevse_esp32_firmware #317: UL issue - WiFi LED

chris1howell, 2022-02-18:
When a station is configured without an LCD Display, station state is not indicated when the station is in standalone mode. Once the station is joined to a WiFi network the LEDs correctly indicate state.

Proposed solution: keep the first LED1 as is, indicating WiFi status. LED2 and higher (if installed) should always be colored to indicate station state regardless of WiFi configuration. LED1 - WiFi State, LED2+ - Station State.

Note-Station under test had no display. WiFi has 5 LEDs 2 onboard OpenEVSE_WiFi_v1 and 3 additional connected to the LED expansion port.

chris1howell, 2022-05-02:
Pixels added via the expansion port no longer show the station status. The onboard pixels LED 1 and 2 work correctly.

jeremypoulter, 2022-06-04:
@chris1howell this works for me, are you sure you have `NEO_PIXEL_LENGTH` set correctly?

jeremypoulter, 2022-06-26:
Tested on OpwnEVSE WiFi board as well for me

closed 2022-06-26 by jeremypoulter

chris1howell, 2022-06-26:
I am not sure this is resolved. I added 3 additional LEDs to the OpenEVSE
WiFi board and changed the number of LEDs in platform.ini.

LEDs 1 and 2 on WiFi work great but LEDs 3, 4 and 5 on the external board
are dark. All 5 work perfectly with a demo sketch that fades through each
color.

jeremypoulter, 2022-06-26:
Which build l/version are you using? I will double check but looked to work fine with the off the shelf openevse_wifi_v1

chris1howell, 2022-06-27:
Just tried the latest Development build... All is good.

PR #320 (2022-02-24): Added support for restricting the WiFi status to a single pixel
Fixes #317