OpenEVSE/openevse_esp32_firmware #321: MQTT data improvements

sspearing, 2022-02-24:
1. wh is a factored duplicate (x1000) of total_energy.  Redundant?  Kept for legacy applications?
2. "session_energy" is reset to zero at a different time than the "total_energy" is updated.  Total energy is updated when the car is unplugged and session energy is zero'd when the car is plugged in next.  Consequence is that during the period between the unplugging and plugging in for the next charge, the energy of the previous charge is counted in both numbers and the two cannot be summed for accurate running total of energy.  Either "wh" or "total_energy" should update at the same time as the "session_energy" is zero'd.  Perhaps there is a rational for the way it is now that I am not seeing?

sspearing, 2022-02-25:
After looking more, I realize the energy tally is handled by the openevse firmware and not by the ESP32.  Before opening an  issue under the other repository, are there any comments from here regarding a change or leaving as is. 

Some options:
1. leave as is = when relay opens add session to total and when relay closes reset session 
   (-ve: creates a period of time when session is double counted)
2. change = when relay closes, add session to total and reset session  
    (-ve: if unit is depowered/unplugged between charges, the session is lost and not added to the total)
3. change = when relay opens, add session to total and reset session 
    (-ve: the previous session data is zeroed as soon as the session is over so could not be viewed)
4. change = when relay opens, copy session to new variable called "previous", add session to total and then reset session to 0.
    (-ve: new variable and associated changes, would have to change display to show previous session when the relay is open)

None of these seem ideal so maybe status quo and to create an accurate running total, use the charger status to determine when the relay is open and subtract off the session_energy from the total? Or have this calculation done by the ESP32 and broadcast and use the result as "total_energy"?

KipK, 2023-03-27:
fixed by https://github.com/OpenEVSE/ESP32_WiFi_V4.x/commit/0145bc101b196617c9efdfb4891b1e7daf79ef5a

closed 2023-03-31 by glynhudson