OpenEVSE/openevse_esp32_firmware #326: 4.1.2 limits power selections in L2 to 15A

mikebaz, 2022-03-09:
I just downgraded from 4.1.2 to 4.1.1 because somehow, the power dropdown decided to limit itself to 15A as a maximum.  Before doing the update, the power was set to 15A for an overnight

glynhudson, 2022-03-09:
> I just downgraded from 4.1.2 to 4.1.1 because somehow, the power dropdown decided to limit itself to 15A as a maximum. Before doing the update, the power was set to 15A for an overnight charge session, but then I did the update and that was the highest I could go.

Does it still limit to 15A when charging level: L2 is selected?

mikebaz, 2022-03-09:
Yes - I even tried changing from L2 (which is where I normally have it) to L1 (which changed the limit to 12A), then back to L2 (which changed the limit back to 15A).

chris1howell, 2022-03-09:
I can verify this. The maximum in the list is whatever the station is set to. In my case the minimum was 6 and the maximum was 24.

jeremypoulter, 2022-03-09:
I did have a look at this, can you check the RAPI page and see what values are being sent from the controller? I think they are being set in the flash in the EVSE, but this then can't be reverted after the first setting. I need to check, as I thought I was setting them as volatile....

Tifaifai, 2022-03-09:
> 
idem with 28A

chris1howell, 2022-03-09:
Looks fine in RAPI.

No trouble setting higher current
> $SC 40
< $OK 40^04

Min and Max look fine.
> $GC
< $OK 6 80 40 40^1E

/config looks strange
..."max_current_soft":24,"min_current_hard":6,"max_current_hard":24,... 
Have not seen Max hard and soft before.

shadowmite, 2022-03-10:
I see the same issue, was set to 24a before update and now I can’t set it higher.

glynhudson, 2022-03-10:
Interesting, my current was set to 32A before the upgrade and is still showing 32A option now. 

For anyone effected by this, a workaround is to issue RAPI command `$SC 32`. This can be done via the web interface if you enable 'advanced' and 'developer toggles. 

[image]

mikebaz, 2022-03-10:
thank you all for the fast replies, confirmation, and correction to the more accurate details, as well as the workaround.  I've gone back up to 4.1.2 knowing I can work around this for now. 🙂

jeremypoulter, 2022-03-10:
I think I understand what the issue is, the `$SC` command does not operate quite as documented:

https://github.com/lincomatic/open_evse/blob/development/firmware/open_evse/rapi_proc.h#L183-L188
does not match the code
https://github.com/lincomatic/open_evse/blob/development/firmware/open_evse/rapi_proc.cpp#L456-L459

And also I think I miss-understood when the `M` option was supposed to be for anyway, but anyway we use the value saved in the EEPROM, exposed as `max_current_hard`, as the maximum current then manage it via setting the volatile settings. This would work fine for fresh OpenEVSE units that have only been used with the v4.1.2 firmware (which is essentially what I did the work with as I was constantly swapping between the US and EU firmware) but existing units that happened to have had a current configured would fail with this issue. 

Hopefully won't be too much of a problem to resolve the issue.

sspearing, 2022-03-13:
My EVSE was set to 22 A when upgraded and per the above comments that became the new max in the drop down on the webUI.  Using RAPI, I set it back to 32 A.  (SC:   $OK 6 80 32 32^1E)

I just wanted to draw you attention to the pilot readout in the image below.  This may be part of what you found already.
[image]

PR #329 (2022-03-17): Properly handle setting the max current by setting in the value in the EVSE EEPROM rather than the WiFi config
Fixes #326
commit 8a7d7ab1 (2022-03-17): Properly handle setting the max current by setting in the value in the EVSE EEPROM rather than the WiFi config

Fixes #326
closed 2022-03-17 by glynhudson