OpenEVSE/openevse_esp32_firmware #328: FR: Timer web UI enhancement

fhteagle, 2022-03-11:
Currently, the timer can be either set or cleared. I propose a few more buttons / options to be added to this UI area:

1. Charge now once for X hours (X set by drop-down)
2. Ignore timer until full once

After either of the above complete, resume normal timer operations.

Use case for this:
I am on a small regional power company that is heavy on solar production, but I don't have my own solar install on my house just yet. So I have to guess when a good time to charge would be, and setting the timer to late morning through early afternoon is a great default guess. However, there are times when I really do want charging to start or proceed outside of the best time, such as storm coming tomorrow, preheating the battery and cabin from wall power before leaving in winter, etc. As I understand the current web UI, I must clear the timer to enable charging in the above cases. Having exception buttons means I can "one shot" these weird cases without having to clear and then remember to re-program the timer.

Hope this makes sense.

Thanks
- DH

jeremypoulter, 2022-03-12:
This is on the roadmap (in my head), and cleverly hidden in #239 and #225, but essentially a combination of those two tickets would add the various limits to the claims API, which can then be use to implement a boost feature, ie override the timers/solar PV until a particular limit has been reached.

fhteagle, 2022-03-12:
I like these ideas a lot. Having the same function get called by a long press of the physical touch button, a web UI button, MQTT topic, OCPP priority claim, etc to "charge now to 100% regardless of every other eco / timer / etc" makes a lot of sense to me. Maybe double/triple/quadruple rapid press of the physical button could be a "1/2/3 hr addition to timer this one time", again made available on web UI, MQTT, OCPP, etc

I look forward to testing it when it's ready.

fhteagle, 2022-03-12:
So I'm on 4.1.0 software currently, and just discovered the "Start" button under charge options. Not sure why I didn't see this before, maybe I wasn't in a timer driven waiting period? Anyway, the start button does seem to connect the car and allow charging as I was contemplating in suggestion 2 above. I assume this must be manually turned off with the clear button? 

So all that is missing compared to my suggestion is resume timer control after X hours and resume timer control at "full" SOC (or when car begins drawing < 6A)?

fhteagle, 2022-10-13:
I think this can be closed and all efforts tracked in  "New UI" issue : https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/401

closed 2022-10-13 by fhteagle