OpenEVSE/openevse_esp32_firmware #331: Temperature monitoring and throttling on ESP32

chris1howell, 2022-03-17:
Move temperature monitoring and throttling to ESP32 leaving the OpenEVSE controller only monitoring for emergency shutdown.

The temperature monitoring and throttling feature on the OpenEVSE controller is not very granular 50% or 100%, and the notification is in RED and causes many support requests. The ESP32 has much more horsepower and can finely adjust current in 1A increments to get the maximum power possible for a given temperature.

Implement monitoring and control in ESP32 with user adjustable point between disabled or 60C to 70C default enabled 65C (shutdown is 72C). Let WiFi increase/decrease current in 1A increments every 10 seconds or so, to maintain the set temperature. Similar to a PID loop or 3D printer heating element.

Provide a softer message to inform user of throttling while charging at normal color (not RED). Suggestion - Temp 65.0C @ 98%
Provide message on screen after charging to inform of throttling during session. Suggestion. Max throttle 2%

chris1howell, 2023-02-24:
Minimum temperature to begin charging - Additional feature to support very low temperatures and auxiliary heater.

Allow user to enable/disable /set a minimum temperature to begin charging. Range of -20C to 10C. Default at -20C.

chris1howell, 2023-02-24:
@KipK - This seems similar to the limit Engine you just did for limit timers. Do you think this is something you could tackle?

KipK, 2023-02-24:
@chris1howell, I guess this is something that could fit directly in the current shaper module. 
Do we have some simple pid or temporal algo on the table for throttling with temperature ?

chris1howell, 2023-02-24:
The PID algo can likely be very simple. This library may a bit more than we need but may be useful. https://github.com/br3ttb/Arduino-PID-Library/

The OpenEVSE controller currently shuts down at 72C. It will set pilot to 50% if 65C is reached. If the temperature declines it will then resume 100% once the temperature drops below 62C. With the ESP32/shaper controlling more it makes sense to only leave the emergency shutdown handled by the OpenEVSE controller.

Here are some graphs of current and temperature. https://openev.freshdesk.com/en/support/solutions/articles/6000248724-openevse-charges-vehicle-fully-but-displays-temperature-warning

Default temperature to begin throttling and attempt to maintain should be 65, however we may want to allow the user to select from 60C to 70C. Shaper can disable above 71.5C and the OpenEVSE controller can keep the emergency shutdown at 72C.

KipK, 2023-03-22:
hi @chris1howell ,

I was about check that, but I see the ESP32 T° sensor is always way above lcd T) sensor ( it can be at 38°C while it's 10°C outside ) 
I think it takes the heat from the ESP through the PCB ( I have tried to revert the pcb position to let the esp32 chip at the top but it doesn't change much ) 

Here is a graph comparing booth sensor until for unknown reason yet , the t° sensor of my lcd stopped reporting so booth T° data merged to the esp32 one. 

[image]

That happened when I've flashed my openevse module with a custom compiled fw, perhaps I have forgot a flag but I don't see which one.

chris1howell, 2023-03-22:
You are correct, The WiFi sensor (Temp 4) runs warmer due to its own heat.
I ran a test with an external sensor just to verify.

The current version of temperature monitoring/Throttling/shutdown on the
OpenEVSE controller only uses Temp1 (retired RTC chip on LCD) or Temp2
Dedicated sensor on LCD. We could use Temp 4 but we would need to add a
conversion factor to compensate for the additional heat from the WiFi.

I would suggest using Temp 2 as the primary, as this is what the controller
will also use for emergency shut down.

For configurations without a display/temp2, then use Temp 4 with offset.
Note the controller will not emergency shutdown in this configuration so we
may need logic to shut down if temp runs away.

chris1howell, 2023-03-22:
> I'm afraid I won't be able for now to check this, my temp sensor on the
> lcd seems either dead or not detected , I haven't tried yet to fallback the
> fw ) . Will have to figure out this first.

Send me a PM to chris {dot} howell at openevse {dot} com , I will send you
some extra hardware for development.

KipK, 2023-03-23:
I got my sensor back, it was probably deactivated on the fw, after making a new one

closed 2026-09-21 by github-actions[bot]