OpenEVSE/openevse_esp32_firmware #332: Charge limit & timer stop/restart Eco Mode

Jgfyyvzss, 2022-03-18:
When in Eco Mode charge Limit & Timer (charge for xx minutes) stop the charge, but on receiving the next Import/Export value charging resumes with no limit or timer.

My expectation is that a "Session" should be from EV connected to EV disconnected, but this does not appear to be the case.

Tested on Firmware OEVSE 7.1.3 & WiFi 4.1.2. Imp/Exp values updated every 60 seconds.

Jgfyyvzss, 2022-03-19:
Confirmed that Limit & Timer work as expected when in Normal mode.

jeremypoulter, 2022-03-20:
These features still use the limits on the EVSE module, need to move those functions to the WiFi module, as discussed in #328

Jgfyyvzss, 2022-03-20:
I had a look at #328, #225 & #239.
It appeared these were talking about adding/improving features rather than the existing feature just not working as expected?
The Limit and Timer both function exactly as expected and wanted, except they reset to zero on receiving the next Export value (or perhaps they reset to zero on completion and then the next Export value simply starts the charge again - to the user it's the same thing).

From my (mis)understanding, adding the features mentioned in the other issues would not make any difference to the fundamental behaviour of the Charge Limit and Timer when in Eco mode.
These both function correctly in Normal mode, it's just when in Eco.

I'm guessing what you say about moving functions to the WiFi module will solve the bug somehow, but just thought I'd check.
I have no functional knowledge of the workings of the software, so excuse me if I'm incorrect here.

PHingamp, 2022-03-23:
Hi,
I confirm the Timer feature doesn't work as expected in Eco mode (I'm using the Eco / PV divert mode to succesfully adapt charger current to available household current usage).

However I've been unable to use the Timer : indeed setting the Start / Stop values then clicking on Set or Update doesn't appear to have any effect on the charge (if the charge is ongoing, it doesn't stop, if charger is waiting, the charge doesn't start). Maybe I'm not using this Timer feature properly ?

In fact the only way I can currently ensure I interrupt the charge before the car reaches 100%, is to disable the Eco mode and use the Time or Energy Limits functionalities which reliably work as expected. But then I'm not adapting current usage to available current.

Would it be possible to offer the Time / Energy Limits also in Eco mode ?

Many thanks for your help !

KipK, 2023-05-16:
this is fixed on 4.2.2 limit engine with new UI

closed 2023-05-16 by KipK