OpenEVSE/openevse_esp32_firmware #333: "Pause status: Disable" not taking effect

casperrijnders, 2022-03-19:
My car (FIAT 500e) requires "Pause status: Disable", because it will not start charging when coming out of sleep (from Timer or PV Divert). A manual Disable and Enable (RAPI $FD $FE) can then trigger it to start charging. 

However when I enable that setting, I don't see any change in behavior. It looks like the setting is not taking effect somehow.

EVSE status via MQTT shows 254 when waiting in Sleep/Pause, regardless if "Pause status: Disable" is set. I would expect it to become 255 (Disabled). Or what would also work is exiting Sleep state by first Disable and then Enable (that would still allow to detect if a vehicle is connected when paused). But none of that is happening.

I upgraded to the latest versions but the issue persists.

/config: "pause_uses_disabled":true

Versions:
"firmware":"7.1.3","protocol":"-","espflash":4194304,"espinfo":"ESP32r1 2 core WiFi BLE BT","buildenv":"openevse_huzzah32","version":"4.1.2"

Am I perhaps missing something?

Thanks,
Casper

jeremypoulter, 2022-03-20:
Looks like this feature got lost in the v4 work, shouldn't be difficult to add back

casperrijnders, 2022-03-22:
Thanks Jeremy, that would be great.

I'd be happy to try out a fix once you have it.

glynhudson, 2022-05-05:
It would be great to get the feature working again, there seems to be a number of new EVs which require a EVSE reboot to wakeup the onboard charger, e.g  Peugeot e-208. Hopefully, disable instead of sleep will allow the car to wakeup

pdhoogh, 2022-05-18:
I have been (one of?) the guy complaining about the Peugeot e-208 not working well together with OpenEVSE on Solar PV Divert. The car goes to sleep after 1 minute of waiting for power and never wakes up again. I discovered that when power is available, resetting the EVSE (not the Wifi) helps initiating the charge when power becomes available again.

So this is the - I'll admit rough handed - python code I'm running to patch this issue. I'm looking forward for the wifi release that fixes the issue in this thread. Currently running 4.1.3.
[image]

pdhoogh, 2022-05-19:
Dank je wel, Casper! 2015 Fiat 500e, cool! I'm on SpeakEV, I wont hijack this forum for EV talk, but it explains why you have the same issue.

I chose for the reset because I noticed some other odd behaviors that seem to be sometimes helped with doing reset. Like charge current modulation no longer working and stuck at max current even if the excess power is not there to do it. Or using a different max current than the one showing or being set in the GUI. Others have reported it too. It is being worked. And it seems to happen more often as I'm doing more tinkering, it does not occur that much when I'm being nice to it. But EVSE Reset isn't always enough. I came across a situation that killed the subscription to the MQTT topics, that required a WiFi reset.

I had another situation twice. The charge LED near the car's charge port turned red. "Error". That only is helped with pulling and reinserting the type 2 plug. I have not seen that since I have the code above running,

closed 2022-06-22 by glynhudson

casperrijnders, 2022-06-22:
Just to confirm, I verified the fix and it does work, using today's development build. 

Now with "pause_uses_disabled":true configured, when pausing the charging, I get "state":255 (disabled) instead of "state":254 (sleep) at at http://x.x.x.x/status 

Also the car resumes charging nicely after clearing the pause.

Thanks a lot @jeremypoulter / @glynhudson !

pdhoogh, 2022-08-10:
Running 4.1.4 for three days now and stopped my "patch" python code _"EVSEDisabler"_ for this issue. 

OpenEVSE seems to do well so far. Pause status set to "Disable". Charge option "Eco", no timers, also not in the Peugeot e208. The car has been plugged in for three days and two nights in a row. In the evening the charge stops. In the morning, as soon as there is enough sunlight the charge resumes. As shown by the trend graphics in MQTT Explorer for Power1 and amp. 

Thanks to @jeremypoulter / @glynhudson !

jeremypoulter, 2022-08-10:
Good to hear, thanks for the feedback