OpenEVSE/openevse_esp32_firmware #334: Teslamate/MQTT Time to full charge

VasDeferens, 2022-03-20:
Using Teslamate and it's MQTT integration I have setup my OpenEVSE to display vehicle data but the Time to Full Charge isn't correct. Teslamate pushes this data as a decimal. Using Tesla API it appears correct but as I use the OpenEVSE to charge another brand of EV in the future I would prefer to reduce waking the Tesla when not needed. Any ideas on how to resolve?

jeremypoulter, 2022-03-20:
The value received over MQTT should be the seconds remaining, is it a different scale?

VasDeferens, 2022-03-21:
Looks like Teslamate reports 2 hours as 2.00

VasDeferens, 2022-03-25:
update: If I use the Tesla API the range is displayed in miles only. Might need a switch to allow kms...

closed 2023-05-16 by KipK