OpenEVSE/openevse_esp32_firmware #335: OCPP 1.6 meterValues incorrect data type

Finalgamer14, 2022-03-25:
I have noticed that the request value does not follow the OCPP 1.6 documentation. We are using 4.1.2 WiFi Firmware.

When the charge point (in this case OpenEVSE wifi module) sends meterValues request meterValue > sampledValue > value must be a string, for it to be compliant with OCPP 1.6.

Request data from charge point:
`
ERROR:ocpp:Error while handling request '<Call - unique_id=531531541, action=MeterValues, payload={'connectorId': 1, 'meterValue': [{'timestamp': '2022-03-25T07:59:21.000Z', 'sampledValue': [{'value': 0, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}, {'timestamp': '2022-03-25T08:00:21.000Z', 'sampledValue': [{'value': 0, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}, {'timestamp': '2022-03-25T08:01:21.000Z', 'sampledValue': [{'value': 0, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}, {'timestamp': '2022-03-25T08:02:21.000Z', 'sampledValue': [{'value': 0, 'measurand': 'Energy.Active.Import.Register', 'unit': 'Wh'}, {'value': 0, 'measurand': 'Power.Active.Import', 'unit': 'W'}]}]}>'
`

SampledValue [OCPP 1.6 documentation](https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf) on page 111

matth-x, 2022-03-28:
@Finalgamer14 Thank you for the report! This issue has already been addressed in the newest development version. Please have a little patience until the fix makes it into the newest release. If you have set up a development environment for OpenEVSE, you can compile the firmware from my development fork with the OCPP fix: https://github.com/matth-x/ESP32_WiFi_V4.x

Actually, a Home Assistant user has already reached out to me with the same problem: https://github.com/lbbrhzn/ocpp/issues/210. If you use Home Assistant as well by any chance, you can follow the overall progress of the integration tests there.

Finalgamer14, 2022-03-28:
Thank you for your answer.
We are creating some modular hardware to extend functionality of OCPP 1.6 compliant charging stations. So I'll try using your development fork of the firmware.

Finalgamer14, 2022-03-28:
I have compiled and flashed your fork of the ESP32 firmware, now the data is correctly validate. For development and prototyping this should be good enough. I'll flash the official release when it comes out.

matth-x, 2022-03-30:
@Finalgamer14 Amazing. If you find any other problem, please continue this issue. If the newest version works fine with your OCPP backend, I would be happy about a feedback as well so I can track the overall compatibility of the OCPP library.

closed 2022-03-30 by matth-x