OpenEVSE/openevse_esp32_firmware #338: Home Assistant for solar power management instead of Solar Power Divert

andriuskr, 2022-03-28:
Hello,

While waiting for solar panel to arrive, starting to think about excess power management for home water heating, cooling/heating and car charging.
While i am total noob about solar, my thought may be wrong, but:

For car charger to use only Solar PV Diver can be not the most efficient way. Because depending on time or home occupancy there may be the priority to make hot water at home instead, cooling/heating  or charging the car, depending on SoC and other factors like scheduled trips for tomorrow. 

Seems logic to use Home Assistant for everything. While there is a very nice integration by [firstof9](https://github.com/firstof9) it still seems there is a big gap on OpenEVSE automation to use only excess solar power or am i missing something here ?

jeremypoulter, 2022-03-28:
Hi @andriuskr, maybe you can clarify what is missing? I haven't looked at the extension in a while but looked like it had the ability to control the charge current and/or enabling 'Eco' mode. If you wanted Home Assistant to manage the divert then I think that is all you would need.

andriuskr, 2022-04-24:
Yes, actually you are right. With good integration from firstof9, i can control charging rate with Home Assistant without any problems.

closed 2022-04-24 by andriuskr

mikebaz, 2022-06-03:
@andriuskr I realize this is closed and wasn't an issue with OpenEVSE per se, but you didn't happen to write-up what you're doing somewhere, did you?  because I'm very curious about getting solar divert working through HA as well but I'm not completely sure how all the pieces fit together let alone how to make it work.