OpenEVSE/openevse_esp32_firmware #347: Max current not followed

doppiaemme, 2022-04-28:
8.2.0.T2
4.1.2

Once I connect the plug, no matter what is the max current setting, it starts charging at 24A and switches to Eco from Normal automatically.

In the example below, despite a limit of 11A it charges at 24A

[image]

I started having this issue since upgrading to one of the latest release for WiFi.
upgrading to 8.2.0 didn't fix it.

doppiaemme, 2022-04-28:
Perhaps..... It seems 24A is about what you would get with the grid excess of 5.9KW that I have in the screenshot, however, it was set to normal and after plugging in the plug it switched to Eco-Divert. 
Why would it switch on its own?

CarvellWakeman, 2022-04-28:
I have a similar, potentially related issue. My max current was previously settable to 60amps, or maybe 50amps? Something higher than 32a. Now the max available is 32a.
[image]

Let me know if this should be a separate issue. Here's my repro steps, but I haven't tested it again yet.
1. OpenEvse advanced kit, assembled, on V4.1.1
2. <Max current is adjustable past 32amps>
3. Set max current to 32amps
4. Update Wifi module to V4.1.2 (latest)

Expected: Max current is still 32amps, but can be set beyond this
Actual: Max current selectable is now limited to 32amps. It may be limited to whatever value was set in V4.1.1.

I saw lots of changes to max current and the evse config in V4.1.2.

doppiaemme, 2022-04-29:
I tried with Safari, Chrome, Brave.... exactly same behaviour

`{"firmware":"8.2.0.T2","protocol":"-","espflash":4194304,"espinfo":"ESP32r1 2 core WiFi BLE BT","buildenv":"openevse_wifi_v1","version":"4.1.2","evse_serial":"Y82140FF061123","wifi_serial":"246F28CD8DD8","mqtt_supported_protocols":["mqtt","mqtts"],"http_supported_protocols":["http"],"diode_check":true,"gfci_check":true,"ground_check":true,"relay_check":true,"vent_check":true,"temp_check":true,"service":2,"scale":220,"offset":0,"max_current_soft":10,"min_current_hard":6,"max_current_hard":24,"ssid":"xxxxxx","pass":"_DUMMY_PASSWORD","www_username":"","www_password":"","hostname":"openevse-8dd8","sntp_hostname":"pool.ntp.org","time_zone":"Europe/Rome|CET-1CEST,M3.5.0,M10.5.0/3","emoncms_server":"https://data.openevse.com/emoncms","emoncms_node":"","emoncms_apikey":"","emoncms_fingerprint":"","mqtt_server":"192.168.3.39","mqtt_port":1883,"mqtt_topic":"openevse","mqtt_user":"","mqtt_pass":"","mqtt_solar":"","mqtt_grid_ie":"homeassistant/sensor/residuo_da_rete_now/state","mqtt_vrms":"homeassistant/sensor/shellyem_b9ea36_channel_1_voltage/state","mqtt_vehicle_soc":"homeassistant/sensor/ax420yx_battery_level/state","mqtt_vehicle_range":"homeassistant/sensor/ax420yx_battery_autonomy/state","mqtt_vehicle_eta":"","mqtt_announce_topic":"openevse/announce/8dd8","ocpp_server":"","ocpp_chargeBoxId":"","ocpp_idTag":"","tx_start_point":"tx_pending","ohm":"","divert_PV_ratio":1.1,"divert_attack_smoothing_factor":0.4,"divert_decay_smoothing_factor":0.05,"divert_min_charge_time":600,"tesla_access_token":"","tesla_refresh_token":"","tesla_created_at":0,"tesla_expires_in":0,"tesla_vehicle_id":"","led_brightness":128,"flags":522,"emoncms_enabled":false,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":true,"pause_uses_disabled":false,"mqtt_vehicle_range_miles":false,"ocpp_enabled":false,"ocpp_suspend_evse":false,"ocpp_energize_plug":false,"mqtt_protocol":"mqtt"}`

chris1howell, 2022-04-29:
> ","emoncms_node":"","emoncms_apikey":"","emoncms_fingerprint":"","mqtt_server":"192.168.3.39","mqtt_port":1883,"mqtt_topic":"openevse","mqtt_user":"","mqtt_pass":"","mqtt_solar":"","mqtt_grid_ie":"homeassistant/sensor/residuo_da_rete_now/state","mqtt_vrms":"homeassistant/sensor/shellyem_b9ea36_channel_1_voltage/state","mqtt_vehicle_soc":"homeassistant/sensor/ax420yx_battery_level/state","mqtt_vehicle_range":"homeassistant/sensor/ax420yx_battery_autonomy/state","mqtt_vehicle_eta":"","mqtt_announce_topic":"openevse/announce/8dd8","ocpp_server":"","ocpp_chargeBoxId":"","ocpp_idTag":"","tx_start_point":"tx_pending","ohm":"","divert_PV_ratio":1.1,"divert_attack_smoothing_factor":0.4,"divert_decay_smoothing_factor":0.05,"divert_min_charge_time":600,"tesla_access_token":"","tesla_refresh_token":"","tesla_created_at":0,"tesla_expires_in":0,"tesla_vehicle_id":"","led_brightness":128,"flags":522,"emoncms_enabled":false,"mqtt_enabled":true,"mqtt_reject_unauthorized":true,"ohm_enabled":false,"sntp_enabled":true,"tesla_enabled":false,"divert_enabled":true,"pause_uses_disabled":false,"mqtt_vehicle_range_miles":false,"ocpp_enabled":false,"ocpp_suspend_evse":false,"ocpp_energize_plug":false,"mqtt_protocol":"mqtt"}

The latest development version fixes this issue.

https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/326

doppiaemme, 2022-05-03:
@chris1howell  do you mean this fix will be included in the next release or there's a pre-release I can already use?
as I tried the .bin under development build (https://github.com/OpenEVSE/ESP32_WiFi_V4.x/releases/download/latest/openevse_wifi_v1.bin)
and it does not fix the issue of auto-switching to Eco mode when the page is refreshed.

doppiaemme, 2022-05-03:
Ok,  I confirm it fixes the max current.
But I was showing another issue, I posted a video above.
This issue is still there and happens on multiple devices.
Whenever I refresh the web page it switches to Eco mode, no matter if it was on Normal

doppiaemme, 2022-05-11:
I have edited the topic...
I think the issue with not saving the setting change from Norm to Eco is related to in general how settings are saved in the GUI.
The same thing happens when in Services->Solar PV divert the check box is ticked and I untick it.
This unticking action doesn't get saved and the only way to save it is to click Save on another section (for instance in the OCPP section).

[image]

Can it be the same issue on the main page? when I switch from Normal to Eco it doesn't get saved?

glynhudson, 2022-05-12:
This issues thread is getting confusing. Please open a new issue ticket for a separate issue. 

I recommend clearing browser cache and testing with a different web browser.

closed 2022-05-12 by glynhudson