OpenEVSE/openevse_esp32_firmware #348: Max Current in V4.1.2 limited by previously configured value in V4.1.1

CarvellWakeman, 2022-04-28:
similar, potentially related issue. My max current could previously be set to 60amps, or maybe 50amps? Something higher than 32a. Now the max available is 32a.
[image]

Hardware: OpenEvse advanced kit, assembled, running V4.1.1 on the ESP32.
Preconditions: <Max current is adjustable past 32amps>

Here's my repro steps, I've performed this operation and reproduced the issue.
1. Set max current to 38amps (or any value > 32amp)
2. Update Wifi module to V4.1.2 (latest)
3. After the page refreshes, attempt to set the max current

Expected 1: Max current is still 38amps, or whatever it was set to on V4.1.1
Expected 2: Max current can be changed to a value higher than what it was on V4.1.1

Actual 1: Max current is set to 32amps after updating to V4.1.2.
Actual 2: Max current cannot be increased past whatever its value was on V4.1.1 before the update.

I saw lots of changes to max current and the evse config in V4.1.2, seems like a bug was introduced.

Edit 1: I can confirm rolling back to V4.1.1 fixes the issue.
[image]

Edit 2: I can confirm rolling forward to V4.1.2 a second time re-introduces the problem.

CarvellWakeman, 2022-04-28:
I have not deployed or tested anything locally, but I've been flipping back and forth from V4.1.1 code and V4.1.2 code.

Here's my unsubstantiated guess. Somehow max_hardware_current is being set to max_configured_current at some point before or during the upgrade. When displaying the list in the UI, the highest value is max_hardware_current.

Due to that, if you have the max configured current set (in the UI) to 38amps for example, then you upgrade firmware, max_hardware_current becomes 38amps somehow.

glynhudson, 2022-04-29:
32A is the maximum permissible current per phase in Europe

CarvellWakeman, 2022-04-29:
This configuration setting (max_hardware_current) limits total current, not current per phase.

CarvellWakeman, 2022-04-29:
I have updated the main post to hopefully make the problem more clear. Let me know if I can provide any more details, such as a video of the issue.

CarvellWakeman, 2022-05-04:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/326

Confirmed this fix in the development build fixes this issue. I did try searching the issues before making this one I promise!

Closing this issue and waiting for the next release.