OpenEVSE/openevse_esp32_firmware #350: Default time schedule and random delay UK legislation

glynhudson, 2022-04-30:
[New UK legislation](https://www.legislation.gov.uk/uksi/2021/1467/contents/made) which is taking effect from June 2022 stipulates that all EVSEs should come pre-configured by default to charge outside of peal hours. “peak hours” means 8am to 11am and 4pm to 10pm.

The charge point should also be set by default to turn on with a randomised delay of up to 10min. if the timer delay is set to switch on at 10pm the random delay will mean it should switch on randomly at some time from 9:50pm - 10:10pm. 

The user should be able to override the timer and the delay if immediate charging is required. The user can opt out from timed charging if they wish. 

To comply with legislation, I propose we preset a time charge on the EVSE between 10pm-8am and create a 10min randomised delay as mentioned above. The user should be able to switch off the random delay via the UI and override the timer and delay using a physical push button.

jeremypoulter, 2022-05-01:
That is an interesting requirement, kinda makes sense. Don't think it will be too difficult.

glynhudson, 2022-05-03:
Indeed, but I can understand the rational. The gov are keen to try to persuade EV owners to charge at off-peak times. Having all new EVSEs to have off-peak timers switched on by default will help with this. The randomised delay is there to avoid having a large load hitting the grid exactly at the same time. 

All this makes sense and in the long run will save everyone money, user are free to opt out of all of this if they would prefer to charge immediately.

Tifaifai, 2022-05-04:
After thinking about it, I understand the logic.
For the physical push button, it is perhaps the occasion to think about the integration of a RFID / NFC reader to be the entry point of authorizations and / or forced triggering of a recharge. (Little kit ;) )

jeremypoulter, 2022-06-04:
@chris1howell will this be an OK default for the USA? @glynhudson Same question for other EU countries?

@Tifaifai Initial RFID support has been merged to the main branch

jketcham, 2022-06-04:
I think if perhaps during initial setup of the EVSE, users could be presented with the default schedule and customize it (or disable it) that would be nice. Or at least make it very obvious how to change the default schedule.

My power company in Phoenix, AZ has an opt-in EV charging program with cheaper rates from 11pm to 5am year round, but no stipulation on a random delay, although I think that's a good idea too.

zymurgic, 2022-07-04:
Note that the GB Legislation (doesn't apply in Northern Ireland) means that in the even of power failure, if it was supposed to be waiting before power failure, when the power returns it shouldn't start charging regardless of the timer or random delay. I assume that's to avoid overloading the local grid after a power failure.

zymurgic, 2022-07-07:
Another clarification (based on how other chargepoint manufacturers are interpreting the regulations) : If you have local generation available, the timed random delays and waiting for off-peak should only apply to energy drawn from the grid. If for instance, you have sufficient local solar for 10A of charging, then it could start using that immediately, then ramp up to full rate once the random delay/ off-peak timer is up.

jeremypoulter, 2022-07-07:
We are only applying the random delay to the schedule, the assumption being that this will be used to charge from the grid. PV divert will charge not have a random delay as it is assumed that this is used for local generation.

jeremypoulter, 2022-07-13:
Default should be 12:30 to 7:30