OpenEVSE/openevse_esp32_firmware #359: HomeAssistant MQTT discovery

jeremypoulter, 2022-05-19:
It probably won't be too much effort to support Home Assistant's MQTT discovery to allow basic setup in Home Assistant even though #246should allow a more complete intergration

glynhudson, 2023-03-30:
This is good HA support via https://github.com/firstof9/openevse

ve6rah, 2023-03-30:
No offense, but that solution is not at all what this issue was about. Home assistant has a protocol for MQTT discovery. The MQTT device publishes information to a specific topic for MQTT discovery. It should be really easy to implement, and then no additional components would need to be installed.

glynhudson, 2023-03-30:
I'm aware of HA MQTT discover, but the OpenEVSE component already exists and is under active development. The component already has all the functionality of MQTT discover plus a lot more. I've also heard that HA are discouraging the use of MQTT discover in favour of dedicated components. Installing a component only take a few clicks and is very painless, for most users it's probably easier than setting up an MQTT server.  

[image]

KipK, 2023-05-03:
Reopening it, as it could be good to have it implemented even if there's another integration.

omriasta, 2023-07-27:
I also think this is a great idea. I have 3 openevse devices. The first at home can use the integration but the other 2 are remote soni have them setup using mqtt. I don't think there is any other way to setup remote devices unless you create a VPN which isn't always possible at the remote site.

closed 2026-09-21 by github-actions[bot]