OpenEVSE/openevse_esp32_firmware #368: collect2.exe: error: ld returned 1 exit status *** [.pio\build\openevse_wifi_v1\firmware.elf] Error 1

DmitryAVR, 2022-06-03:
I don't understand what is wrong, how can I fix it?

c:/users/d.averkin/.platformio/packages/toolchain-xtensa32@2.50200.97/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.project.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\openevse_wifi_v1\firmware.elf] Error 1

[image]

jeremypoulter, 2022-06-03:
Can you give the full log and command you are running?

DmitryAVR, 2022-06-03:
**It is platformio.ini**

[platformio]
data_dir = src/data
default_envs = openevse_wifi_v1

[common]
version = -D BUILD_TAG=4.1.3
monitor_speed = 115200
lib_deps =
  bblanchon/ArduinoJson@6.19.1
  jeremypoulter/ArduinoMongoose@0.0.18
  jeremypoulter/Micro Debug@0.0.5
  jeremypoulter/ConfigJson@0.0.4
  jeremypoulter/OpenEVSE@0.0.11
  jeremypoulter/ESPAL@0.0.3
  jeremypoulter/StreamSpy@0.0.1
  jeremypoulter/MicroTasks@0.0.2
  erropix/ESP32 AnalogWrite@0.2
  lorol/LittleFS_esp32@1.0.5
  matth-x/ArduinoOcpp@0.1.0
lib_ignore = WebSockets ; ArduinoOcpp: don't compile built-in WS library
extra_scripts = scripts/extra_script.py
debug_flags =
  -ggdb
  -D ENABLE_DEBUG
  #-D ENABLE_DEBUG_WEB
  #-D ENABLE_DEBUG_WEB_REQUEST
  #-D ENABLE_DEBUG_SCHEDULER
  #-D ENABLE_DEBUG_EVSE_MAN
  #-D ENABLE_DEBUG_EVSE_MONITOR
  #-D ENABLE_DEBUG_DIVERT
  #-D ENABLE_DEBUG_LED
  #-D ENABLE_DEBUG_LCD
  #-D ENABLE_DEBUG_MQTT
  #-D ENABLE_DEBUG_EMONCMS
  #-D ENABLE_DEBUG_TESLA_CLIENT
  #-D ENABLE_PROFILE
src_build_flags =
#  -D ARDUINOJSON_USE_LONG_LONG
# -D ENABLE_ASYNC_WIFI_SCAN
build_flags =
  -D CS_PLATFORM=CS_P_ESP32
  -D MG_ENABLE_SSL=1
  -D MG_ENABLE_HTTP_STREAMING_MULTIPART=1
  -D MG_SSL_MBED_DUMMY_RANDOM=1
  -D MG_SSL_IF=MG_SSL_IF_MBEDTLS
  -D MG_SSL_IF_MBEDTLS_FREE_CERTS=1
  #-D MG_SSL_IF_MBEDTLS_MAX_FRAG_LEN=2048
  #-D ARDUINO_MONGOOSE_DEFAULT_STREAM_BUFFER=2048
  #-D ARDUINO_MONGOOSE_SEND_BUFFER_SIZE=2048
  #-D ENABLE_DEBUG
  #-D CS_ENABLE_DEBUG
  #-D MBEDTLS_DEBUG_C
  -D MG_ENABLE_SNTP=1
  -D CS_PLATFORM=CS_P_ESP32
  -D AO_CUSTOM_WS ; ArduinoOcpp: don't use built-in WS library
  -D AO_CUSTOM_CONSOLE ; ArduinoOcpp: use custom debug out
  -D AO_DBG_LEVEL=AO_DL_WARN
  #-D ENABLE_DEBUG
  #-D ENABLE_DEBUG_MONGOOSE_HTTP_CLIENT
  -D RAPI_MAX_COMMANDS=20
  -D BUILD_ENV_NAME="$PIOENV"
build_partitions = min_spiffs.csv
monitor_flags = --filter=esp32_exception_decoder

neopixel_lib = adafruit/Adafruit NeoPixel@1.7.0

# specify exact Arduino ESP SDK version, requires platformio 3.5+ (curently dev version)
# http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform
[... 980 lines cut ...]
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\main.cpp.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\openevse_wifi_v1\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\openevse_wifi_v1\libFrameworkArduino.a
Linking .pio\build\openevse_wifi_v1\firmware.elf
c:/users/d.averkin/.platformio/packages/toolchain-xtensa32@2.50200.97/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.project.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\openevse_wifi_v1\firmware.elf] Error 1
======================================================================================== [FAILED] Took 215.25 seconds ========================================================================================

Environment       Status    Duration
----------------  --------  ------------
openevse_wifi_v1  FAILED    00:03:35.255
==================================================================================== 1 failed, 0 succeeded in 00:03:35.255 ====================================================================================

DmitryAVR, 2022-06-03:
in the middle of this log there is such an error. Visual photo:

[image]

jeremypoulter, 2022-06-26:
This should have been resolved as part of #207

closed 2022-06-26 by jeremypoulter