OpenEVSE/openevse_esp32_firmware #370: Limit available charge rate to 48A maximum

mikebaz, 2022-06-03:
Are there any OpenEVSE designs that go above 48A at this time?  The OpenEVSE web site only lists 40A and 48A max units.  I'm asking because if there isn't, shouldn't the max limit be 48A as well (not 80A)?

This might be a dumb issue/question.

glynhudson, 2022-06-04:
This is a question for @chris1howell, in Europe we're limited to 32A max. I'm pretty sure the J1772 connector can't handle much more than about 40A.

chris1howell, 2022-06-04:
Many customers use OpenEVSE controllers at 80A for J1772.

There is a one time command to set the maximum station hardware.  We set
this at the time of manufacture

The command to set a station to 48A maximum is $SC 48 M

The M will set the maximum in EEPROM one time and can not be changed
without first erasing the EEPROM.

mikebaz, 2022-07-27:
its sounds like this should be closed so I'm going to close it :)

closed 2022-07-27 by mikebaz