OpenEVSE/openevse_esp32_firmware #373: MQTT manual modes , also Reorganising MQTT topics

KipK, 2022-06-06:
Let's discuss about [ https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/369#discussion_r890166699 ] here, as it involve bigger changes.

So when MQTT set something that is not a configuration data, it should be a claim, that could be considered on mqtt side as such "manual_override" , let's call it "manual" then, to not be confusing with manual_override from the api ( using the highest priority manual level ).

/manual[0,1]				(0: auto/no manual 1: either manual charge_current or status are active)
/manual/set 0               (0: delete all related claims, can't set 1, there's no purpose I think?)
/manual/charge_current      (0: auto/no claim, other value in Amp )
/manual/charge_current/set  ( set to 0 delete the charge_current claim, other value in Amp )
/manual/status 				( 0: auto/no claim, 1: enabled/start, 2: disabled/sleep )
/manual/status/set			( set to 0 delete the enabled/disabled claim, 1: enabled/start, 2: disabled/sleep )

Considering this, we could also reorganize a bit the mqtt topics and remove duplicated values.
 Something like that ? 
 
/announce
/announce/[id]

/state [0,1,...]
/state/colour [0,1,..]
/state/flags [int]
/state/vehicule [0,1]
/state/manual_override [0,1]

/smartmode [0,1]  are smart features enabled or not ( could be solar divert / Grid +i-E  / other incoming modes]
/smartmode/set [0,1] enable/disable smart features. Replace divertmode/set( could be solar divert / Grid +i-E  / other incoming modes]
/smartmode/divert_update
/smartmode_grid_ie
/smartmode/available_current
/smartmode/smoothed_available_current
/smartmode/charge_rate ( should be deleted, just rounded available_current, confusing with pilot ) 

/charge/pilot
/charge/current
/charge/session_kwh
/charge/total_kwh

/system/temp
/system/tempwifi
/system/voltage
/system/max_current
/system/max_current/set
/system/srssi
/system/freeram

KipK, 2022-06-07:
You join me on this one, I was exploring this possibility too:

override ( list overrides in json )
/override/set {json doc | toggle | delete }

/claim ( claim from mqtt client id in json )
/claim/set [json doc] update claim
/claim/set delete (delete claim)

/schedule ( get all schedules ) 
/schedule/set ( batch update schedule )

/status ( get status json): For this one I'm not sure we should export the full json as it would need a full refresh for just one value update.

So there should we organise a bit those data perhaps between /status /system /config ?

KipK, 2022-06-07:
Also I would rename the enable/disable divert mode setting to something more generic in preparation to new smart features. ( /smart_mode enable/disable ? )
For now I use divertmode in grid mode without solar panel to handle charge load balancing and it works well, but I plan to make a dedicated service with some added features.

KipK, 2022-06-07:
I have started to integrate that. There's something that tickle me in the way mqtt is handled in the current code.

OpenEVSE Mqtt client  disconnect from the server and reconnect to update the whole data, in a loop.
There's no need to do that. We can publish the data with "retain=true", and publish them allinone only once at startup.
Then we just need to update retained values when it's needed using event only. Considering we have the MQTT Last Man Standing, other clients knows OpenEVSE is disconnected and can ignore those values if it's offline.

KipK, 2022-06-07:
I've pushed a WIP on this branch based on ispidfv4: https://github.com/KipK/ESP32_WiFi_V4.x/commit/d9979dd18a95e9ccda0d8777304f45270707d693

jeremypoulter, 2022-06-11:
Sorry for not getting back sooner.

I think I would prefer to keep the manual API fairly basic, it really should be limited to a users direct intervention rather than any automated process so keeping it fairly basic would encourage use of the claims for all the more advanced use cases.

I think the claim bits are fine, I would perhaps use 'clear' or 'release' over 'delete' as this is more natural. `DELETE` is only used for the HTTP API as that is the defined verb in HTTP.

On nesting the topics, I think this will have a lot of compatibility implications, eg won't work with EmonCMS via MQTT, and if we keep the same mapping across HTTP/MQTT/WebSockets there will be a lot of code to rework in the GUI and other stuff like the Home Assistant integration. We certainly could look at renaming some topics (as in duplicate them and deprecate the old one), but lets do that as a separate task.

Specifically on renaming divert_mode to smart_mode, I don't think it makes sense to do this, PV diversion is only one 'smart' mode there are others, eg Ohm, and will be  more that are not mutually exclusive so bundling turning them on/off under the same control doesn't make sense. If you need to just switch them all on/off for some reason, that is what the manual override is for.

Hopefully this doesn't sound too negative, your contribution is very much appreciated and very good work so far.

KipK, 2022-06-11:
no problem with your comments.  I've used Delete matching the http api to keep the main logic only, let's use "clear" "relase", whatever ;)

about https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/373#issuecomment-1148491231 , I'm trying to fire events on mqtt for claims change/add/release. I see the HTTP interface receiving those updates in real time, but not on MQTT. But http page is polling for that. 
Should I do the same for MQTT, or shouldn't we try to add that to current event engine and make it more generic like other data updates ?

KipK, 2022-06-26:
Closed by https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/379

PR #379 (2022-07-29): Mqtt reworked API version
Fixes #358 
Fixes #376 
Fixes #373 

Following previous discussion here https://github.com/OpenEVSE/ESP32_WiFi_V4.x/pull/369 
Based on espidf4 branch.
 
Edit: all functional
closed 2022-06-26 by KipK