OpenEVSE/openevse_esp32_firmware #374: New Wifi kit missing temperature sensor

omriasta, 2022-06-06:
I have 3 older OpenEVSE chargers and have upgraded the chip in 2 of the units without issue.
I just installed the wifi v1 module in the last and noticed that the temp sensor is missing.
The only difference I noticed is that the device is labeled:
`ESP32r2 2 core WiFi BLE BT`
and on my other units I have:
`ESP32r1 2 core WiFi BLE BT`

I have also read elsewhere that the internal temp sensor on newer esp32 units is being "removed". I am not sure if this is the issue I have or if I need to do something to enable it? Or possibly just a defective unit?
I also checked the json returned from the api and it gives me `"temp4":false`

Does anyone know what the case is here or have any troubleshooting advice?

omriasta, 2022-06-06:
Yes they are all on the same version now, yes config is openevse_wifi_v1.
I received it with 4.1.2 and it was missing, just updated to 4.1.3 and still missing.
Is this info not applicable:
[https://github.com/espressif/arduino-esp32/issues/2422](https://github.com/espressif/arduino-esp32/issues/2422)

jeremypoulter, 2022-06-06:
We don't use the internal temp sensor for that very reason, it is an external chip that is used.

omriasta, 2022-06-07:
That makes sense...ok, opened a ticket with support...maybe it is just a defective unit...

omriasta, 2022-06-07:
Just spoke to support, apparently the sensors are not available and therefore the modules are now shipping without it.

closed 2022-06-07 by omriasta

omriasta, 2022-06-13:
For anyone else, I ordered the recommended part and installed it in 2 minutes, super easy and recommended.
Thank you for the quick response!